The-OpenROAD-Project/OpenROAD #679: TritonRoute GR errors out

rovinski, 2021-05-25:
Not providing a guide file to `detailed_route` causes TritonRoute GR to run, but it errors out.

OpenROAD version:
```
OpenROAD 1 89c42b436a0ecfdb0420b89c3bbc4425920a242c
```

System info:
```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
```

Backtrace:
```
Program received signal SIGSEGV, Segmentation fault.
0x0000555556312d0b in std::__cxx11::list<fr::frNode*, std::allocator<fr::frNode*> >::begin (this=0x48)
    at /usr/include/c++/9/bits/stl_list.h:942
942           { return iterator(this->_M_impl._M_node._M_next); }

#0  0x0000555556312d0b in std::__cxx11::list<fr::frNode*, std::allocator<fr::frNode*> >::begin (this=0x48)
    at /usr/include/c++/9/bits/stl_list.h:942
#1  0x0000555556306d6d in fr::FlexGR::initGR_initObj_net (this=0x7fffffffd720, net=0x55555a3888b0)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/gr/FlexGR.cpp:1449
#2  0x0000555556306b64 in fr::FlexGR::initGR_initObj (this=0x7fffffffd720)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/gr/FlexGR.cpp:1427
#3  0x0000555556304194 in fr::FlexGR::initGR (this=0x7fffffffd720)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/gr/FlexGR.cpp:967
#4  0x00005555562fe28e in fr::FlexGR::main (this=0x7fffffffd720, db=0x555558adb8e8)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/gr/FlexGR.cpp:57
#5  0x00005555562e8e15 in triton_route::TritonRoute::gr (this=0x555558a5c550)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/TritonRoute.cpp:198
#6  0x00005555562e918f in triton_route::TritonRoute::main (this=0x555558a5c550)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/TritonRoute.cpp:231
#7  0x00005555562fb5d4 in detailed_route_cmd (guideFile=0x7ffff7fac028 "", outputGuideFile=0x7ffff7fac028 "",
    outputMazeFile=0x7ffff7fac028 "", outputDrcFile=0x7ffff7fac028 "", outputCmapFile=0x7ffff7fac028 "",
    dbProcessNode=0x7ffff7fac028 "", drouteEndIter=-1, drouteViaInPinBottomLayerNum=-1,
    drouteViaInPinTopLayerNum=-1, orSeed=-1, orK=0, bottomRoutingLayer=0x7ffff7fac028 "",
    topRoutingLayer=0x7ffff7fac028 "", verbose=1)
    at OpenROAD-flow-scripts/tools/build/OpenROAD/src/TritonRoute/CMakeFiles/TritonRoute.dir/TritonRouteTCL_wrap.cxx:1744
#8  0x00005555562fc6c2 in _wrap_detailed_route_cmd__SWIG_0 (clientData=0x0, interp=0x555558a978f0, objc=15,
    objv=0x555558b1cf80)
    at OpenROAD-flow-scripts/tools/build/OpenROAD/src/TritonRoute/CMakeFiles/TritonRo--
ute.dir/TritonRouteTCL_wrap.cxx:2038
#9  0x00005555562fd0be in _wrap_detailed_route_cmd (clientData=0x0, interp=0x555558a978f0, objc=15,
    objv=0x555558b1cf80)
    at OpenROAD-flow-scripts/tools/build/OpenROAD/src/TritonRoute/CMakeFiles/TritonRoute.dir/TritonRouteTCL_wrap.cxx:2179
#10 0x00007ffff7e2c5f2 in TclNRRunCallbacks () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#11 0x00007ffff7ecb694 in Tcl_RecordAndEvalObj () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#12 0x00007ffff7ef058d in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#13 0x000055555566c4a4 in main (argc=1, argv=0x7fffffffe238)
    at OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:224
```

testcase: 
[detail_route_gcd_nangate45_base_2021-05-25_17-14.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6542417/detail_route_gcd_nangate45_base_2021-05-25_17-14.tar.gz)

QuantamHD, 2021-05-25:
Is there some potential benefit to the Triton Route global router, or do we think that fastroute will provide better results?

I ask because I think it's potentially splitting resources by having two global routers that might be better spent on a single global router. @maliberty

maliberty, 2021-05-25:
The claim from BX was that WXL is better is some ways.  It shares the pin access code with TR which is one current advantage. 
 @eder-matheus did some study but in general it has been languishing.  I integrated it so that it wouldn't become impossible later as the code bases diverged.  However it has no unit tests so it seems something has gone wrong.  If it is easy to fix I would do so but I'm not committed to it (nor totally ready to give up either).

maliberty, 2021-05-25:
You can lookup the "TritonRoute-WXL: The Open Source Router with Integrated DRC Engine" paper for more details.

vijayank88, 2023-06-14:
@rovinski 
With latest commit not facing any segfault.
```
[INFO DRT-0198] Complete detail routing.
Total wire length = 5693 um.
Total wire length on LAYER metal1 = 15 um.
Total wire length on LAYER metal2 = 2418 um.
Total wire length on LAYER metal3 = 2766 um.
Total wire length on LAYER metal4 = 463 um.
Total wire length on LAYER metal5 = 0 um.
Total wire length on LAYER metal6 = 29 um.
Total wire length on LAYER metal7 = 0 um.
Total wire length on LAYER metal8 = 0 um.
Total wire length on LAYER metal9 = 0 um.
Total wire length on LAYER metal10 = 0 um.
Total number of vias = 2616.
Up-via summary (total 2616):.

---------------
 active       0
 metal1    1307
 metal2    1242
 metal3      63
 metal4       2
 metal5       2
 metal6       0
 metal7       0
 metal8       0
 metal9       0
---------------
           2616

[INFO DRT-0267] cpu time = 00:00:07, elapsed time = 00:00:05, memory = 688.79 (MB), peak = 688.79 (MB)

[INFO DRT-0180] Post processing.
```
Can we close this?

closed 2023-06-14 by maliberty

rovinski, 2023-06-14:
@vijayank88 did you actually run TritonRoute GR? The issue was not with not providing a guide file (which since this issue was posted, guides are now being passed in ODB anyways), the issue was with TritonRoute GR segfaulting which is what happens when DRT is invoked and no guides exist.

If we want to just mark GR as "won't fix" and close the issue, that's fine too. But the code still exists in the repository and is still callable AFAICT.