The-OpenROAD-Project/OpenROAD #685: Documentation on using the Python SWIG interface for things like OpenDB?

donn, 2021-05-27:
According to @mithro this project has official Python SWIG support now, but there appears to be no documentation for it and/or any way to build and use it. Can someone in the know weigh in?

maliberty, 2021-05-27:
We are deprecating standalone opendb's python executable in favor of "openroad -python".  The later has the same database apis as opendb but it doesn't support all the same commands as the tcl interface to openroad does.

donn, 2021-05-27:
Ah, I have no interest in the executable per se. The app I'm building (https://github.com/cloud-v/dffram) uses a Python API, i.e.:

```python3
import opendbpy as odb

db = odb.dbDatabase.create()
db = odb.read_lef(db, "/path/to/lef")
db = odb.read_lef(db, "/path/to/tlef")
```

Is there a way to make post-standalone OpenDB OpenROAD behave similarly? I have an odd Frankenstein I'm maintaining that couples some of the old standalone OpenDB code to achieve this.

maliberty, 2021-05-27:
could your python code run inside openroad?

donn, 2021-05-27:
I would have to assess what running "inside openroad" entails first (any pointer to documentation for that would be appreciated) but I am not opposed to the idea.

maliberty, 2021-05-27:
Unfortunately there isn't any documentation yet (nor was there any for opendb python).  Basically it would replace your python interpreter so openroad -python ... instead of python ....

We could also look at exporting just a python library like we used to for loading into a standard interpreter if needed.

donn, 2021-05-27:
Let me toy around with `openroad -python` and report back, then.

QuantamHD, 2021-05-27:
It's not too bad @donn I have the import module working in bazel. You basically get a .so and a .py out of the make file swig module in src/OpenDB. You then read it that .py file in the normal way, and it will attempt to dynamically link that .so file in the same directory it exists in.

donn, 2021-05-27:
@QuantamHD Mind sharing your build commands?

donn, 2021-05-28:
NVM- Cmake worked fine if you just "import opendb". I'm gonna test if this works overall… Just that there's some Python 3.8 syntax that needs to be unexisted in my project first.

donn, 2021-05-28:
Alright, that appears to have worked fairly simply: no need to mess with shared objects or anything, just `openroad -python` suffices. You can then just `import opendb` and everything works as Opendbpy previously did.

Though another gotcha is the locale is set to ascii for some reason so you'd need to set these environment flags: `-e LC_ALL=en_US.UTF-8 -e LANG=en_US.UTF-8`.

But overall, yeah. I can probably put https://github.com/Cloud-V/Opendbpy to rest now. Thanks y'all

QuantamHD, 2021-05-29:
@donn can we consider this issue closed? I'll make another issue to update the documentation on the openroad -python command.

donn, 2021-05-29:
Yeah, that'll be it.

closed 2021-05-29 by donn