The-OpenROAD-Project/OpenROAD #686: Unable to install OpenROAD using clone and build from sources

nishita2626, 2021-05-28:
Hi -

I am using Ubuntu 20.04 in WSL (Windows Subsystem for Linux). I am trying to install OpenROAD using clone and build and getting error. 

[https://openroad.readthedocs.io/en/latest/user/GettingStarted.html](url)

$ git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow
$ cd OpenROAD-flow
$ ./build_openroad.sh --local

I have installed all the install dependencies tools and libraries given here [https://github.com/The-OpenROAD-Project/OpenROAD](url)

Getting the error. See image -

[image]

Please help. 

Thanks,
Nishita

maliberty, 2021-05-28:
I think the docs need a bit of updating.  Please run etc/DependencyInstaller.sh -dev which should get you the needed Qt library.

QuantamHD, 2021-05-28:
To add some more context I believe the issue is related to this question https://askubuntu.com/questions/1034313/ubuntu-18-4-libqt5core-so-5-cannot-open-shared-object-file-no-such-file-or-dir

With the solution being to run the strip command `sudo strip --remove-section=.note.ABI-tag /usr/lib64/libQt5Core.so.5`

Which if you follow @maliberty's advice will be run as part of etc/DependencyInstaller.sh -dev

nishita2626, 2021-05-28:
> I think the docs need a bit of updating. Please run etc/DependencyInstaller.sh -dev which should get you the needed Qt library.

Hey I'm getting this message after executing - 

-bash: etc/DependencyInstaller.sh: No such file or directory

maliberty, 2021-05-29:
Are you running it in your OpenROAD directory?  You might try ./etc/DependencyInstaller.sh in case you don't have the current directory in your PATH.

nishita2626, 2021-05-29:
> Are you running it in your OpenROAD directory? You might try ./etc/DependencyInstaller.sh in case you don't have the current directory in your PATH.

Please check this image. 

[image]

maliberty, 2021-05-29:
You need to cd into tools/OpenROAD

nishita2626, 2021-05-29:
There is no such directory named tools/OpenROAD

maliberty, 2021-05-29:
Then you forgot to do 'git submodule update --init --recursive'

nishita2626, 2021-05-29:
> Then you forgot to do 'git submodule update --init --recursive'

Yes I had forgotten but even after executing this I'm getting the same error. Kindly help. 

[image]

maliberty, 2021-05-29:
You need to run as root, you can use sudo

nishita2626, 2021-05-30:
Thank you so much! OpenROAD has been installed successfully. You've been a great help. Thanks again.

maliberty, 2021-05-30:
Glag you succeeded.

commit 4b5043ee (2021-08-22): closes #686
commit 53e701ad (2021-08-22): Merge pull request #1021 from The-OpenROAD-Project-staging/db-memory-leak

closes #686
closed 2021-05-30 by maliberty