The-OpenROAD-Project/OpenROAD #690: Why isn't diamond search height an input?

donn, 2021-05-31:
While updating OpenLane to use a relatively recent OR commit, I was met with this:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/b762efa78105daa31935d6d18345f86680a91543/src/dpl/src/Opendp.cpp#L132

There's been a patch in the OpenLane repo forever that allows this number to be provided as an optional input. but naturally the codebase has changed significantly since then.  Is there any reason why this hasn't been made an input by now? Should I open a PR?

QuantamHD, 2021-05-31:
This is my non-authoratative opinion.

I think a PR here would be nice with caveat that options in general are counter to the projects goals.

OpenROAD's goal is to simplify the IC design process with full automation. The existing EDA tools are completely overrun with options that make it extremely difficult to get a useable result out of them.

I think a PR to add this as an option would be best if it also included code that could automatically set it based on the design.

maliberty, 2021-05-31:
Please explain why it needs to be an option?  Is there a specific issue that can't be resolved without it?  We did make a change to handle instances placed on macros automatically which was the only reason for needing to increase the limit that I know of.

donn, 2021-06-01:
Then I believe that settles that.  Apologies, I'm just taking a deep dive into OpenLane's OpenROAD scripts and I'm trying to make sense of some of the inconsistencies.

closed 2021-06-01 by donn