The-OpenROAD-Project/OpenROAD #691: Significance of RSZ-0022

donn, 2021-05-31:
Potential stupid quesstion incoming:

```
OpenROAD 1 b762efa78105daa31935d6d18345f86680a91543
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openLANE_flow/designs/inverter/runs/openlane_test/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 440 library cells
[INFO ODB-0226] Finished LEF file:  /openLANE_flow/designs/inverter/runs/openlane_test/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openLANE_flow/designs/inverter/runs/openlane_test/tmp/floorplan/6-pdn.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 2 pins.
[INFO ODB-0131]     Created 32 components and 116 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 0 connections.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: /openLANE_flow/designs/inverter/runs/openlane_test/tmp/floorplan/6-pdn.def
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 28520 43520
[INFO GPL-0006] NumInstances: 32
[INFO GPL-0007] NumPlaceInstances: 1
[INFO GPL-0008] NumFixedInstances: 31
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 2
[INFO GPL-0011] NumPins: 4
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 34165 54885
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 28520 43520
[INFO GPL-0016] CoreArea: 750720000
[INFO GPL-0017] NonPlaceInstsArea: 98844800
[INFO GPL-0018] PlaceInstsArea: 3753600
[INFO GPL-0019] Util(%): 0.58
[INFO GPL-0020] StdInstsArea: 3753600
[INFO GPL-0021] MacroInstsArea: 0
[INFO GPL-0031] FillerInit: NumGCells: 130
[INFO GPL-0032] FillerInit: NumGNets: 2
[INFO GPL-0033] FillerInit: NumGPins: 4
[INFO GPL-0023] TargetDensity: 0.75
[INFO GPL-0024] AveragePlaceInstArea: 3753600
[INFO GPL-0025] IdealBinArea: 5004800
[INFO GPL-0026] IdealBinCnt: 150
[INFO GPL-0027] TotalBinArea: 750720000
[INFO GPL-0028] BinCnt: 8 8
[INFO GPL-0029] BinSize: 2875 4080
[INFO GPL-0030] NumBins: 64
[ERROR RSZ-0022] no buffers found.
Error: or_replace.tcl, 92 RSZ-0022
```
I should quite hope there are no buffers considering that there aren't any in the design, which is a smoke-test single inverter. What am I doing wrong here?

QuantamHD, 2021-05-31:
In general we need a reproducible test case to address issues. This is usually in the form of the TCL script used to run the command and its required files.

However, It looks like the line in question is complaining that resizer was run and it couldn't find any buffers in your cell library i.e. liberty file. https://github.com/The-OpenROAD-Project/OpenROAD/blob/ba7cb6557354a3280959e8122f184da866703be5/src/rsz/src/Resizer.cc#L404

donn, 2021-05-31:
Well... there simply wasn't a liberty file loaded. We didn't need one for older versions, curiously. Wonder when that changed.

Thanks.

closed 2021-05-31 by donn