The-OpenROAD-Project/OpenROAD #695: Regression: key "VPWR" not known in dictionary

donn, 2021-06-01:
Another oddity while updating the OpenROAD version on OpenLane:

```
OpenROAD 1 b762efa78105daa31935d6d18345f86680a91543
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 440 library cells
[INFO ODB-0226] Finished LEF file:  merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: inverter.floorplan.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 2 pins.
[INFO ODB-0131]     Created 32 components and 116 component-terminals.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: inverter.floorplan.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: common_pdn.tcl
[INFO PDN-0008] Design Name is inverter
[INFO PDN-0009] Reading technology data
[INFO PDN-0011] ****** INFO ******
Type: stdcell, grid
    Stdcell Rails
      Layer: met1  -  width: 0.480  pitch: 2.720  offset: 0.000 
    Straps
      Layer: met4  -  width: 1.600  pitch: 7.708  offset: 3.854 
      Layer: met5  -  width: 1.600  pitch: 11.042  offset: 5.521 
    Connect: {met4 met5} {met1 met4}
Type: macro, macro_1
    Macro orientation: R0 R180 MX MY R90 R270 MXR90 MYR90
    Straps
    Connect: {met4_PIN_ver met5}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - grid
[INFO PDN-0015] Writing to database
-code 1 -level 0 -errorcode NONE -errorinfo {key "VPWR" not known in dictionary
    while executing
"dict get $global_connections $net_name"
    (procedure "get_valid_mterms" line 5)
    invoked from within
"get_valid_mterms $net_name"
    (procedure "get_global_connect_list_default" line 8)
    invoked from within
"get_global_connect_list_default [dict get $design_data core_domain] false"
    (procedure "export_opendb_global_connection" line 8)
    invoked from within
"export_opendb_global_connection"
    (procedure "export_opendb_specialnets" line 13)
    invoked from within
"export_opendb_specialnets"
    (procedure "opendb_update_grid" line 4)
    invoked from within
"opendb_update_grid"
    (procedure "apply" line 7)
    invoked from within
"apply $config"
    (procedure "pdngen::apply_pdn" line 13)
    invoked from within
"pdngen::apply_pdn $config_file $verbose "} -errorline 9
[ERROR PDN-9999] Unexpected error: key "VPWR" not known in dictionary
```

Here are the relevant files. Please note that they're tarred, xzipped and gzipped, in that order, to be uploadable to GitHub (Simple .gz compression does a horrendous job of compressing the liberty file and .xz files cannot be uploaded to GitHub.)

[packaged.tar.xz.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6574665/packaged.tar.xz.gz)

QuantamHD, 2021-06-01:
This seems similar to #633. It seems there is some kind of naming issue with the pdn config.

donn, 2021-06-04:
Any ETA on a fix? I'm blocked on this.

osamahammad21, 2021-06-07:
@maliberty  depends on [OR-1015](https://openroad.atlassian.net/browse/OR-1015?atlOrigin=eyJpIjoiZDExMmI5M2Q5MTYzNDliYjg1NzVjMjk0ZTc2NjEwMmQiLCJwIjoiaiJ9)

donn, 2021-06-07:
Ah, met with the JIRA blackhole. Welp. Is there any workaround at least? Problem is when I just tried renaming the pins to VDD and VSS other facilities appeared to break.

maliberty, 2021-06-07:
Jira summary: the suggested solution leads to a crash so there is no solution or workaround yet.

donn, 2021-06-07:
Heck. Well, keep me apprised.

maliberty, 2021-06-09:
Fwiw this was the suggestion:
[image]

donn, 2021-06-09:
Eh, I'll give it a shot.

ibrahimkhairy, 2021-06-10:
OK, I'll have a look

osamahammad21, 2021-06-21:
@donn power pins are generated now as instructed in [the comment](https://github.com/The-OpenROAD-Project/OpenROAD/issues/695#issuecomment-858010283) . Also you will need to update pdngen::global_connections with the power and ground names. For openlane, just set FP_PDN_ENABLE_GLOBAL_CONNECTIONS to 1

donn, 2021-06-21:
Will try & report

donn, 2021-06-24:
Oh yeah this is fixed now. Thanks for the help.

closed 2021-06-24 by donn