The-OpenROAD-Project/OpenROAD #700: Unable to build PDNSim

nishita2626, 2021-06-02:
Hi -

I have installed all the pre-requisites mentioned here [https://github.com/The-OpenROAD-Project/PDNSim](url) but I am unable to build it. Could you please let me know how do I solve this error?

[image]

Thanks.

maliberty, 2021-06-02:
The standalone repo is archived as pdnsim has been integrated to openroad.  I suggest just using openroad but if you required the standalone version then @VidyaChhabria is the author and could help.

nishita2626, 2021-06-02:
Okay, thanks!

VidyaChhabria, 2021-06-02:
I would agree with @maliberty here.  The latest of pdnsim is integrated into OpenROAD and the standalone is not being developed anymore so I suggest using openroad. The related commands for PDNSim can be found at the far end of the [README ](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/README.md) and a simple test to run PDNSim is [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/psm/test/gcd_no_vsrc.tcl) after you have an openroad binary.

closed 2021-06-02 by maliberty