The-OpenROAD-Project/OpenROAD #702: could not connect to x display

AyushOza07, 2021-06-02:
hello, I am using centos 8 for openroad and facing error will using command openroad  - gui

bash-4.2$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.

maliberty, 2021-06-02:
Can you run any X application?  It sounds like you don't have DISPLAY set.

AyushOza07, 2021-06-03:
> Can you run any X application?  It sounds like you don't have DISPLAY set.

yeah i am able to open x application.
i was running waylant before but now i have switched to x11 server in which i am able to open x application like klayout but openroad is not able to connect to x server

i can see x server running by systemctl

maliberty, 2021-06-03:
What platform are you on?

AyushOza07, 2021-06-03:
> What platform are you on?
Centos 8.3 aarch64

QuantamHD, 2021-06-03:
Are you running openroad in docker?

If so running this command might fix it `xhost +local:docker`

maliberty, 2021-06-03:
I don't have access to an arm based machine to test.  I'm guessing you need to set QT_QPA_PLATFORM  but I'm not certain what the right setting would be.  See https://doc.qt.io/qt-5/embedded-linux.html  It looks like it is try xcb.  It looks like wayland is supported too: https://www.qt.io/blog/2018/05/29/whats-new-in-qt-5-11-for-the-wayland-platform-plugin

AyushOza07, 2021-06-03:
> Are you running openroad in docker?
> 
> If so running this command might fix it `xhost +local:docker`

yes i am running operoad in docker

QuantamHD, 2021-06-03:
You need to run that command in the host session, so outside docker in bash.

mithro, 2021-06-03:
https://hdl.github.io/containers/#_tools_with_gui

mithro, 2021-06-03:
The page at https://hdl.github.io/containers/#_tools_with_gui has a lot of information about running GUI apps in containers.

AyushOza07, 2021-06-03:
> Are you running openroad in docker?
> 
> If so running this command might fix it `xhost +local:docker`

$ xhost +local:docker
non-network local connections being added to access control list

after this i again tried to run command openroad -gui but same problem occurred again

bash-4.2$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.

maliberty, 2021-06-03:
As a quick experiment try "xhost +" which will disable all access controls and determine whether this is the issue or not.  'xhost -' will restrict access again.

vvbandeira, 2021-06-14:
Hi, I was able to get gui applications working inside docker with
```
xhost +local:docker
docker run -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=unix:${DISPLAY} [...]
```
I used a VNC session with Centos7, but should work for others

closed 2021-06-15 by AyushOza07