The-OpenROAD-Project/OpenROAD #708: Unable to build OpenROAD. Error code 1

PiyushSaini97, 2021-06-05:
Hi -

I am using Ubuntu 20.04 and have been trying to install OpenROAD in the system. I've already cloned it in my home directory using the following -
$git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
$cd OpenROAD-flow
$./build_openroad.sh --local

At the final step I encounter error. See image -
[image]

I have also attached the Makefile in the image which supposedly resulted into error. Already installed all the prerequisite dependencies mentioned here https://github.com/The-OpenROAD-Project/OpenROAD

Please Help.

Thanks,
Piyush

maliberty, 2021-06-05:
That's odd as there is no obvious error.  If you run the strip command yourself do you get an error or a non-zero exit code (eg echo  $? in bash right afterwards)?

PiyushSaini97, 2021-06-05:
It returns non-zero (in my case it is **2**) exit code.

maliberty, 2021-06-05:
That sounds like no such file - does the file you are trying to strip exist?

Were there any other errors earlier in the build process?

PiyushSaini97, 2021-06-05:
No errors. Just encountered some warnings while cloning OpenROAD.

maliberty, 2021-06-05:
Something is not right.  Please do a new checkout and build and capture the full output.

PiyushSaini97, 2021-06-05:
> That sounds like no such file - does the file you are trying to strip exist?
> 

I removed the files and reinstalled everything again. Also, updated the dependencies using the command **./etc/DependencyInstaller.sh** which I referred from here :
https://github.com/The-OpenROAD-Project/OpenROAD/issues/686

It worked for me this time. Maybe one of the file was missing while cloning. 

Thank you for your help.

closed 2021-06-06 by maliberty