The-OpenROAD-Project/OpenROAD #713: make: *** No rule to make target 'platforms//config.mk.'

nishita2626, 2021-06-08:
Hi -
I have added a new design in Makefile, vedic8x8 and have commented the default design. See image

[image]

This is the config.mk file for vedic multiplier in the OpenROAD-flow/flow/designs/nangate45/vedic8x8

[image]

After writing the make command I am getting error

[image]

Please help. Thanks

maliberty, 2021-06-08:
The 'Permission denied' is likely the root of the problem.  Can you do

cat designs/nangate45/vedic8x8/config.mk

and see if the file is readable?

nishita2626, 2021-06-08:
It shows permission denied.

maliberty, 2021-06-08:
Sounds like you have a basic unix file permission issue.  Try chmod u+r designs/nangate45/vedic8x8/config.mk

nishita2626, 2021-06-08:
chmod: cannot access 'designs/nangate45/vedic8x8/config.mk': Permission denied

maliberty, 2021-06-08:
This really a file permission issue rather than an openroad issue that you'll need to resolve at your side.  ls -l designs/nangate45/vedic8x8/config.mk is your starting point to see what's going on.

closed 2021-06-08 by maliberty