The-OpenROAD-Project/OpenROAD #719: Question: how to specify clock nets in verilog code

nishita2626, 2021-06-10:
Hello-

I have created a new design and have made all appropriate files and folders to store the results. After writing _make_ command, I am getting the error that no clock nets have been found.

[image]

 
Could you please let me know what changes are to be made in my code for resolving this issue? I have attached my code herewith. Thanks.

**Code for 8 bit vedic multiplier:**

`module half_adder(input a,b, output sum, carry);

	assign sum = a ^ b;
	assign carry = a & b;

endmodule
module full_adder(input a, b, cin, output sum, cout);

	wire sum1,carry1,carry2;
	half_adder HA0(a,b,sum1,carry1);
	half_adder HA1(cin,sum1,sum,carry2);

	assign cout = carry1 | carry2;

endmodule
module ripple_adder_4bit(input [3:0] a,b, input cin, output [3:0] sum, output cout);

	wire carry1, carry2, carry3;
	full_adder FA0(a[0],b[0],cin,sum[0],carry1);
	full_adder FA1(a[1],b[1],carry1,sum[1],carry2);
	full_adder FA2(a[2],b[2],carry2,sum[2],carry3);
	full_adder FA3(a[3],b[3],carry3,sum[3],cout);

endmodule
module ripple_adder_6bit(input [5:0] a,b, input cin, output [5:0] sum, output cout);

	wire carry1, carry2, carry3, carry4, carry5;

	full_adder FA0(a[0],b[0],cin,sum[0],carry1);
	full_adder FA1(a[1],b[1],carry1,sum[1],carry2);
	full_adder FA2(a[2],b[2],carry2,sum[2],carry3);
	full_adder FA3(a[3],b[3],carry3,sum[3],carry4);
	full_adder FA4(a[4],b[4],carry4,sum[4],carry5);
	full_adder FA5(a[5],b[5],carry5,sum[5],cout);

endmodule
module ripple_adder_8bit(input [7:0] a,b, input cin, output [7:0] sum, output cout);

	wire carry;

	ripple_adder_4bit RA0(a[3:0],b[3:0],cin,sum[3:0],carry);
	ripple_adder_4bit RA1(a[7:4],b[7:4],carry,sum[7:4],cout);

endmodule
module ripple_adder_12bit(input [11:0] a,b, input cin, output [11:0] sum, output cout);

	wire carry;

	ripple_adder_6bit RA0(a[5:0],b[5:0],cin,sum[5:0],carry);
	ripple_adder_6bit RA1(a[11:6],b[11:6],carry,sum[11:6],cout);

endmodule
module vedic2x2(input [1:0] a,b, output [3:0] prod);

	wire a1b1 = a[1] & b[1];
	wire a0b1 = a[0] & b[1];
	wire a1b0 = a[1] & b[0];
	wire a0b0 = a[0] & b[0];
	wire carry;
	
	assign prod[0] = a0b0;

	half_adder HA0(a0b1,a1b0,prod[1],carry);
	half_adder HA1(a1b1,carry,prod[2],prod[3]);

endmodule
module vedic4x4(input [3:0] a,b, output [7:0] prod);

	wire [3:0] mult0, mult1, mult2, mult3;
	wire [3:0] sum0;
	wire [5:0] sum1, sum2;
	wire carry0, carry1, carry2;

	vedic2x2 VD0(a[1:0],b[1:0],mult0);
	vedic2x2 VD1(a[1:0],b[3:2],mult1);
	vedic2x2 VD2(a[3:2],b[1:0],mult2);
	vedic2x2 VD3(a[3:2],b[3:2],mult3);

	ripple_adder_4bit RA0({2'b0,mult0[3:2]},mult2,1'b0,sum0,carry0);
	ripple_adder_6bit RA1({2'b0,mult1},{mult3,2'b0},1'b0,sum1,carry1);
	ripple_adder_6bit RA2({2'b0,sum0},sum1,1'b0,sum2,carry2);
	
	assign prod = {sum2,mult0[1:0]};

endmodule
module vedic8x8(input [7:0] a,b, output [15:0] prod);

	wire [7:0] mult0, mult1, mult2, mult3;
	wire [7:0] sum0;
	wire [11:0] sum1, sum2;
	wire carry0, carry2, carry3;

	vedic4x4 VD0(a[3:0],b[3:0],mult0);
	vedic4x4 VD1(a[3:0],b[7:4],mult1);
	vedic4x4 VD2(a[7:4],b[3:0],mult2);
	vedic4x4 VD3(a[7:4],b[7:4],mult3);

	ripple_adder_8bit RA0({4'b0,mult0[7:4]},mult2,1'b0,sum0,carry0);
	ripple_adder_12bit RA1({4'b0,mult1},{mult3,4'b0},1'b0,sum1,carry1);
	ripple_adder_12bit RA2({4'b0,sum0},sum1,1'b0,sum2,carry2);

	assign prod = {sum2,mult0[3:0]};

endmodule
module test;

	reg [7:0] a,b;
	wire [15:0] prod;
	reg clk;

	vedic8x8 U0(a,b,prod);

	initial begin
		clk = 0;
		#500;
		$finish;
	end

	always@(posedge clk) begin
		a = $random;
		b = $random;
		#20;
	end

	always #10 clk = !clk;

	reg [7:0] a_reg,b_reg;

	always@(posedge clk) begin
		a_reg <= a;
		b_reg <= b;
	end

	always@(posedge clk)begin
		if((a_reg > 0) && (b_reg > 0)) begin
			if(prod == a_reg *b_reg)
				$display("%d x %d = %d, Test Passed", a_reg, b_reg, prod);
			else
				$display("%d x %d = %d, Test Failed", a_reg, b_reg, prod);
		end
	end

endmodule`

maliberty, 2021-06-10:
@mathursanjiv I think this should be a warning not an error and CTS should just return as there is no work to do.

You could work around this by hacking up flow/scripts/cts.tcl to just skip cts and repair_clock_nets

rovinski, 2021-06-11:
This is similar to #299

maliberty, 2021-06-11:
The more things change, the more they stay the same.  Probably this cropped up due to the logger conversion.

nishita2626, 2021-06-11:
> This is similar to #299

So is there some report I need to update?

maliberty, 2021-06-11:
No its clear what the problem is from this report.

nishita2626, 2021-06-11:
I mean what is the next step I should follow?

maliberty, 2021-06-11:
You can either wait for a fix or use the workaround I suggested earlier:

You could work around this by hacking up flow/scripts/cts.tcl to just skip cts and repair_clock_nets

closed 2021-06-14 by mathursanjiv