The-OpenROAD-Project/OpenROAD #735: Pins Not Being Properly Exported

donn, 2021-06-14:
OR commit: 1dae3ba766284714256c53813a117f3a0062279c

PDNGEN does not appear to export VDD and VSS as pins, rather, only as specialnets. Not really sure what's going on here.

[or_upgrade_test_or_pdn_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6648050/or_upgrade_test_or_pdn_packaged.tar.gz)

donn, 2021-06-16:
Potentially helpful (or useless info): I once was re-running the scripts haphazardly while updating the or_issue script and it appears to have generated pins. Not sure what the exact repro steps here, but thought I should mention it.

donn, 2021-06-21:
Apparently, fix is best suited for `open_pdks`? https://github.com/osamahammad21/open_pdks/commit/5aa6367857b2209bd186a9cad5735894f782fb6c

@osamahammad21 Can you confirm?

osamahammad21, 2021-06-23:
@donn yes, you have to specify the pins metal layer in the pdn config. I will close that now since the solution is known.

closed 2021-06-23 by osamahammad21