The-OpenROAD-Project/OpenROAD #761: TritonCTS `runTritonCTS` segment failed in gtest

zekungao, 2021-06-18:
Excuse me, I am now learning codes of `TritonCTS` and try to use gTest to test it.
While running `runTritonCts()`, core dumped,
I was running same flow with `test/simple_test.tcl`

Here's my codes:

  const char* defName = "src/TritonCTS/test/16sinks.def";
  const char* lefName = "src/TritonCTS/test/Nangate45/Nangate45.lef";
  const char* libName = "src/TritonCTS/test/Nangate45/Nangate45_typ.lib";

  ord::OpenRoad* openRoad;
  Tcl_Interp* interp = Tcl_CreateInterp();
  Tcl_Init(interp);
  openRoad = ord::OpenRoad::openRoad();
  openRoad->getDbNetwork()->clear();
  openRoad->getSta()->clear();
  ord::tclAppInit(interp);

  openRoad->readLef(lefName, "test", true, true);
  openRoad->getSta()->readLiberty(
        libName, openRoad->getSta()->cmdCorner(), sta::MinMaxAll::all(), true);
  openRoad->readDef(defName, false, false, false);
  // give arguments necessary to tritonCTS
  Tcl_Eval(interp, "create_clock -period 5 clk");
  Tcl_Eval(interp, "set_wire_rc -clock -layer metal3");
  TritonCTS* cts = openRoad->getTritonCts();
  cts->getParms()->setRootBuffer("CLKBUF_X3");
  ASSERT_NO_THROW(cts->setBufferList("CLKBUF_X3"));
  cts->getParms()->setSinkBuffer("CLKBUF_X3");
  cts->getParms()->setWireSegmentUnit(20);
  cts->getParms()->setSinkClustering(false);

  cts->runTritonCts(); // <=here segment fail
  openRoad->getDb()->clear(); 

And the result is as following:

    Eden 1 bf0818133c50a401442cb0b8185912bfd142c7b2
  This program is licensed under the BSD-3 license. See the LICENSE file for details.
  Components of this program may be licensed under more restrictive licenses which must be honored.
  [INFO ODB-0222] Reading LEF file: src/TritonCTS/test/Nangate45/Nangate45.lef
  [INFO ODB-0223]     Created 22 technology layers
  [INFO ODB-0224]     Created 27 technology vias
  [INFO ODB-0225]     Created 134 library cells
  [INFO ODB-0226] Finished LEF file:  src/TritonCTS/test/Nangate45/Nangate45.lef
  [INFO ODB-0127] Reading DEF file: src/TritonCTS/test/16sinks.def
  [INFO ODB-0128] Design: test_16_sinks
  [INFO ODB-0130]     Created 1 pins.
  [INFO ODB-0131]     Created 16 components and 96 component-terminals.
  [INFO ODB-0133]     Created 1 nets and 16 connections.
  [INFO ODB-0134] Finished DEF file: src/TritonCTS/test/16sinks.def
   *****************
   * TritonCTS 2.0 *
   *****************
   *****************************
   *  Create characterization  *
   *****************************
  [INFO CTS-0039] Number of created patterns = 2376.
  [INFO CTS-0084] Compiling LUT
  Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
  2           4           1           34          1           26          
  [WARNING CTS-0043] 792 wires are pure wire and no slew degration.
  TritonCTS forced slew degradation on these wires.
  [INFO CTS-0046]     Num wire segments: 2376
  [INFO CTS-0047]     Num keys in characterization LUT: 838
  [INFO CTS-0048]     Actual min input cap: 1
   **********************
   *  Find clock roots  *
   **********************
  [INFO CTS-0002]  User did not specify clock roots.
   ************************
   *  Populate TritonCTS  *
   ************************
   Initializing clock nets
   Looking for clock nets in the design
  [INFO CTS-0007]  Net "clk" found for clock "clk"
  [INFO CTS-0009]  Initializing clock net for : "clk"
  [INFO CTS-0010]  Clock net "clk" has 16 sinks
  [INFO CTS-0008]  TritonCTS found 1 clock nets.
   ****************************
   *  Check characterization  *
   ****************************
      The chacterization used 1 buffer(s) types. All of them are in the loaded DB.
   ***********************
   *  Build clock trees  *
   ***********************
  [INFO CTS-0027]  Generating H-Tree topology for net clk
  [INFO CTS-0028]     Tot. number of sinks: 16
  [INFO CTS-0030]     Number of static layers: 0
    [INFO CTS-0020]  Wire segment unit: 14000  dbu (7 um)
    [INFO CTS-0023]  Original sink region: [(3730, 1730), (22730, 20730)]
    [INFO CTS-0024]  Normalized sink region: [(0.266429, 0.123571), (1.62357, 1.48071)]
    [INFO CTS-0025]     Width:  1.3571
    [INFO CTS-0026]     Height: 1.3571
    [WARNING CTS-0045] Creating fake entries in the LUT.
    Level 1
        Direction: Vertical
        # sinks per sub-region: 8
        Sub-region size: 1.3571 X 0.6786
    [INFO CTS-0034]     Segment length (rounded): 1
        Key: 2376 outSlew: 2 load: 1 length: 1 isBuffered: false
    [INFO CTS-0032]  Stop criterion found. Max number of sinks is (15)
    Building clock sub nets...
    [INFO CTS-0035]  Number of sinks covered: 16
    [INFO CTS-0033]  Clock topology of net "clk" done.
    ****************
    * Post CTS opt *
    ****************
    [INFO CTS-0036]  Avg. source sink dist: 17902.50 dbu.
    [INFO CTS-0037]  Num outlier sinks: 0
    ********************
    * Write data to DB *
    ********************
    Writing clock net "clk" to DB
    [INFO CTS-0018]     Created 3 clock buffers.
    Segmentation fault (core dumped)

While I try `Tcl_EvalFile(interp, "src/TritonCTS/test/simple_test.tcl")`
It gave me same result.
I guess I have missed something to init `openRoad` or `TritonCTS`
But I have called all codes that i can found.

Could you plz give me some advice?
Apologies to give you a stupid question.

maliberty, 2021-06-19:
I'm not clear why you don't just
```
cd cd src/TritonCTS/test/
openroad simple_test.tcl
```
to run this test.  Why write your own driver that is mostly calling tcl_eval?

zekungao, 2021-06-21:
It is because i wanna do unit tests for every functions in TritonCTS,
and in case their functions changed if i modify some codes.

`openroad simple_test.tcl` runs well
but it only do a global regression for TritonCTS.

Is it impossible to call functions individually in `TritonCTS`?
Sry to disturb you with a stupid question

maliberty, 2021-06-21:
No worries I'm just trying to understand so I can best help you.  In your code you are calling cts->runTritonCts which is the top level entry point you would get from calling the tcl command.  Is your goal to have more fine grained tests or just setup different situations that would lead to more coverage.  The later is easier to do and would require less c++ coding but sometimes it is tricky to hit a particular condition.

zekungao, 2021-06-21:
Thank you for your understanding.
Indeed, i'm going to *have more fine grained tests*.
I tried every functions before `writeDateToDb()` individual , for example `buildClockTrees` and `populateTritonCts`
They all run correctly.

I have test all parts of `TritonCTS` in gtest except this one.
If it is difficult to achieve, maybe I should give it up temporarily.

maliberty, 2021-06-22:
Is this a project you are doing with the intention to submit it or just for your own use?  If the former we should discuss how this would integrate as we don't currently use gtest.

zekungao, 2021-06-22:
It is *just for my own use* and not urgent.
I'm now doing this just for learning algorithm and codes of `TritonCTS`.
So if this will cost you too much energy, I don't think there is any need to waste your precious time.
I can just use another way to do the same thing, maybe, gtest is not necessary.
-----------------------------------------------------------------------------------------------------------
I just wonder why after building `HTreeBuilder` successfully, `writeDateToDb()` failed.
If it is possible to run tcl script, why I can't call these function directly in cpp?
Is that I forgot anything before init `TritonCTS`? openRoad? openDB? or Tcl_interp?

These are my questions indeed.
I'm sorry for such a chattering.

maliberty, 2021-06-22:
I don't know off hand why it crashes.  The initialization sequence is a bit involved so you might have missed a step.  You could debug writeDataToDb as a learning exercise as well.

closed 2021-07-12 by zekungao