The-OpenROAD-Project/OpenROAD #774: Global placer SegFaults on floorplan output

quentinw91, 2021-06-22:
Hi,
I am trying to setup a new platform for this flow.  I took the files from the nangate45 example (removed some stuff like the macro definitions), but the flow errors with a segmentation fault after the floorplan.  
```
...

[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 51, enlarge 23, costheight 509, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 29 cost_step 5 L 1 cost_type 1 updatetype 4.
[INFO GRT-0126] Overflow Report:
[INFO GRT-0127] total Usage   : 2731
[INFO GRT-0128] Max H Overflow: 41
[INFO GRT-0129] Max V Overflow: 29
[INFO GRT-0130] Max Overflow  : 41
[INFO GRT-0131] Num Overflow e: 10
[INFO GRT-0132] H   Overflow  : 55
[INFO GRT-0133] V   Overflow  : 44
[INFO GRT-0134] Final Overflow: 99

[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 52, enlarge 23, costheight 519, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 30 cost_step 5 L 1 cost_type 1 updatetype 4.
[INFO GRT-0126] Overflow Report:
[INFO GRT-0127] total Usage   : 2731
[INFO GRT-0128] Max H Overflow: 41
[INFO GRT-0129] Max V Overflow: 29
[INFO GRT-0130] Max Overflow  : 41
[INFO GRT-0131] Num Overflow e: 9
[INFO GRT-0132] H   Overflow  : 56
[INFO GRT-0133] V   Overflow  : 41
[INFO GRT-0134] Final Overflow: 97

[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 53, enlarge 23, costheight 529, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 30 cost_step 5 L 1 cost_type 1 updatetype 4.
[INFO GRT-0126] Overflow Report:
[INFO GRT-0127] total Usage   : 2731
[INFO GRT-0128] Max H Overflow: 41
[INFO GRT-0129] Max V Overflow: 29
[INFO GRT-0130] Max Overflow  : 41
[INFO GRT-0131] Num Overflow e: 11
[INFO GRT-0132] H   Overflow  : 55
[INFO GRT-0133] V   Overflow  : 44
[INFO GRT-0134] Final Overflow: 99

[INFO GRT-0124] Update congestion history type 4.
[INFO GRT-0102] iteration 54, enlarge 23, costheight 539, threshold 0 via cost 0. 
log_coef 0.43634596, healingTrigger 31 cost_step 5 L 1 cost_type 1 updatetype 4.
[WARNING GRT-0164] initial grid wrong y1 x1 [0 0] , net start [0 1] routelen 54.
[WARNING GRT-0167] checking failed 1.
Segmentation fault (core dumped)

```

It can be reproduced with the following testcase:
[20210622.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6695488/20210622.zip)

eder-matheus, 2021-06-22:
@quentinw91 Your script is calling global_router before global_place. I was able to run without errors with this script:

```
# Read lef
read_lef platforms/test_plat/lef/NangateOpenCellLibrary.tech.lef
read_lef platforms/test_plat/lef/NangateOpenCellLibrary.macro.mod.lef

# Read liberty files
read_liberty platforms/test_plat/lib/NangateOpenCellLibrary_typical.lib

# Read design files
read_def results/2_floorplan.def
read_sdc results/2_floorplan.sdc

# Set res and cap
source platforms/test_plat/setRC.tcl

#set_dont_use $::env(DONT_USE_CELLS)

# set fastroute layer reduction
#source platforms/test_plat/fastroute.tcl
set_global_routing_layer_adjustment metal2 0.8
set_global_routing_layer_adjustment metal3 0.7
set_global_routing_layer_adjustment metal4-metal10 0.4

set_routing_layers -signal metal2-metal10
set_macro_extension 2

global_placement -routability_driven -density 0.30 \
    -pad_left 2 \
    -pad_right 2

global_route -guide_file results/route.guide \
             -overflow_iterations 100 \
             -verbose 2

write_def results/3_1_place_gp.def
```

maliberty, 2021-06-22:
@eder-matheus please fix it so you get an error message rather than a segv.

closed 2021-06-22 by eder-matheus

quentinw91, 2021-06-23:
Yeah, sorry, I was trying to reproduce the issue and got confused with the scripts while debugging.  I am also able to run your script, but I still have an issue when I try to run global_place.tcl:
[global_place_gcd_test_plat_base_2021-06-23_07-43.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6699998/global_place_gcd_test_plat_base_2021-06-23_07-43.tar.gz)

eder-matheus, 2021-06-23:
@quentinw91 Looks like you're using an old version of our flow scripts. In your platforms/test_plat/fastroute.tcl, it calls the `global_route` command, while our new scripts don't do it anymore. Check the new fastroute.tcl script here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/nangate45/fastroute.tcl

quentinw91, 2021-06-23:
Oh ok, that's where it was coming from... Thanks !