The-OpenROAD-Project/OpenROAD #792: ERROR GPL-0306 RePlAce diverged at wire/density gradient Sum.

mooredan, 2021-06-24:
This is a very minimal design: two inverter connected together.  Get to global_placement and end up with the error:

global_placement -verbose 3
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 1500 14400
[INFO GPL-0004] CoreAreaLxLy: 0 0
[INFO GPL-0005] CoreAreaUxUy: 13500 14400
[INFO GPL-0006] NumInstances: 2
[INFO GPL-0007] NumPlaceInstances: 2
[INFO GPL-0008] NumFixedInstances: 0
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 3
[INFO GPL-0011] NumPins: 6
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 14697 14697
[INFO GPL-0014] CoreAreaLxLy: 0 0
[INFO GPL-0015] CoreAreaUxUy: 13500 14400
[INFO GPL-0016] CoreArea: 194400000
[INFO GPL-0017] NonPlaceInstsArea: 0
[INFO GPL-0018] PlaceInstsArea: 129600000
[INFO GPL-0019] Util(%): 66.67
[INFO GPL-0020] StdInstsArea: 129600000
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG Error: 0.00000000 HPWL: 65700
[InitialPlace]  Iter: 2 CG Error: 0.00000000 HPWL: 68014
[InitialPlace]  Iter: 3 CG Error: 0.00000000 HPWL: 68244
[InitialPlace]  Iter: 4 CG Error: 0.00000000 HPWL: 68271
[InitialPlace]  Iter: 5 CG Error: 0.00000005 HPWL: 68274
[DEBUG GPL-0003] FillerInit: CoreArea 194400000
[DEBUG GPL-0003] FillerInit: WhiteSpaceArea 194400000
[DEBUG GPL-0003] FillerInit: MovableArea 136080000
[DEBUG GPL-0003] FillerInit: TotalFillerArea 6480000
[DEBUG GPL-0003] FillerInit: NumFillerCells 0
[DEBUG GPL-0003] FillerInit: FillerCellArea 64800000
[DEBUG GPL-0003] FillerInit: FillerCellSize 4500 14400
[INFO GPL-0031] FillerInit: NumGCells: 2
[INFO GPL-0032] FillerInit: NumGNets: 3
[INFO GPL-0033] FillerInit: NumGPins: 6
[INFO GPL-0023] TargetDensity: 0.70
[INFO GPL-0024] AveragePlaceInstArea: 64800000
[INFO GPL-0025] IdealBinArea: 92571432
[INFO GPL-0026] IdealBinCnt: 2
[INFO GPL-0027] TotalBinArea: 194400000
[INFO GPL-0028] BinCnt: 2048 2048
[INFO GPL-0029] BinSize: 7 8
[INFO GPL-0030] NumBins: 4194304
[DEBUG GPL-0003] npinit: InitialHPWL: 24100
[DEBUG GPL-0003] npinit: BaseWireLengthCoef: 0.0333333
[DEBUG GPL-0003] npinit: InitSumOverflow: 40.8482
[DEBUG GPL-0003] updateWireLengthCoef:  NewWireLengthCoef: 0.00333333
[DEBUG GPL-0003] npinit: WireLengthCoef: 0.00333333
[DEBUG GPL-0003] updateGrad:  DensityPenalty: 0
[DEBUG GPL-0003] updateGrad:  WireLengthGradSum: 4.37213
[DEBUG GPL-0003] updateGrad:  DensityGradSum: nan
[DEBUG GPL-0003] updateGrad:  GradSum: nan
[ERROR GPL-0306] RePlAce diverged at wire/density gradient Sum.

Traced the error to back to source file: replace/src/nesterovPlace.cpp, method: NesterovPlace::updateGradients

Minimal flow script too:
set process "amic5n"
read_lef ${process}/${process}.tlef
read_lef ${process}/${process}_std_cell.lef
read_liberty ${process}/${process}_tt.lib
read_verilog simple.v
link_design simple
initialize_floorplan -utilization 60 -site core
make_tracks
auto_place_pins metal2
global_placement -verbose 3

simple.v:
module simple(z, a);
  output z;
  input a;

  wire x1;

  inv_b G1 (
    .z(x1),
    .a(a)
  );

  inv_b G2 (
    .z(z),
    .a(x1)
  );
endmodule

Am I missing a step?   ...no other WARNING messages before the error.

Pointers where to look?  Thank you.

maliberty, 2021-06-25:
Sometimes the placer has issues with very small designs (it reaches a solution so early it thinks something went wrong).  I'll take a look.

mooredan, 2021-06-25:
Here's a self-contained test case to duplicate the issue. 

openroad -exit cmd.tcl

Tested with latest code base. 

Thanks.
[OpenRoad_tc1.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6715337/OpenRoad_tc1.tar.gz)

mooredan, 2021-06-25:
I've done some tracing, and it looks like the nan values are originating from FFT::doFFT()'s call to ddct2d() 
The binDensity_ array is populated with nan

This is during NesterovPlace::init()

Comparing to a larger design, the binDensity_ array does get set with real values during this time

mooredan, 2021-06-25:
...and check out the NumBins: for this minimal design: 4194304     a slightly larger design (a 8 bit gray counter) has 16

QuantamHD, 2021-06-28:
You can disable the initial placement and see if that works as a workaround for this issues

closed 2021-06-28 by maliberty

mooredan, 2021-06-28:
Confirmed that the commit fixed this issue. Thank you.