The-OpenROAD-Project/OpenROAD #798: Simple two cell design, strange routing

mooredan, 2021-06-29:
This is a simple two cell design that I'm using for learning and evaluation -- it's two inverters and two ports:

[image]

Ports and pins are in metal1 -- the routing looks obvious: just three pieces of metal1 and you're done.  However, the router does something strange:

[image]

It's jumping up to metal2 when not necessary (in all cases), and doing some "scenic routing"

Is there a way to direct the router to conserve upper layer resources? 
Complete stand-alone test case provided, just run make to reproduce. 
[OpenRoad_tc2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6731172/OpenRoad_tc2.tar.gz)

Thanks, Dan

maliberty, 2021-06-29:
We avoid routing on m1 as that tends to lead to pin access problems (not in this simple case but in general).  However the scenic routes will be investigated.

Stephanommg, 2021-06-29:
@mooredan the straight wire segments on metal1 didnt happen because there are high routing costs for running over/into the pins, due the the reasons Matt mentioned above.

Still, speaking the net a, it seems it could go metal2, and go straight to the left IO pin, and then go metal1. This didnt happen because TritonRoute forbids via access on IO pins at the moment.

Thus, in these very simple scenarios you can expect to see those things.

maliberty, 2021-06-29:
We generally don't see IOs on m1 so that is unusual.  In other layers planar access works well.  @mooredan is there anything more to pursue here?

mooredan, 2021-06-29:
This isn't a modern process so routing resources are limited with only three layers of metal, and a couple of more complex cells use some metal2.  Just looking at it from the perspective of what would a human do in these situations where a simple and obvious route in metal1 wouldn't occupy those precious metal2 resources.

Looking on the left hand side, it did do the route in metal1 (it did so too on the right hand side, but not as cleanly), and then did the unnecessary, and redundant, m1,via1,m2,via1 route blocking two metal2 tracks (metal2's preferred direction is horizontal).

Thanks for looking into this.

maliberty, 2021-07-06:
Do you think such simple cases occur often?  It can really only happen between adjacent pins.  We have focused on 130nm and below for which this isn't a concern.

It never really routed in m1 so much as the m1 shapes happened to touch in the process of going up and down to m2.

mooredan, 2021-07-06:
Sure, if it's possible and within the design rules, then it can occur.  My concern is needlessly utilizing the upper level routing resource when it isn't necessary to do so.  I did confirm that if the input and output pins were place on metal2 (in the same locations), then a much cleaner route occurs -- however it still is using metal2 for the intermediate route when a small piece of metal1 would suffice.

That said, anyone in the ASIC business knows that we spend significant time learning the tools that we work with and understanding their limitations and devising workarounds.  The tools from the "big three" aren't immune from these sort of things either.  However, I must say that support is so much better with FOSS.

maliberty, 2021-07-06:
It certainly could occur but I expect it will be very low frequency so the actual savings will be small.  I'm not basing this on any hard data and am open to being proven wrong on a more realistic design.

maliberty, 2021-07-20:
I'm going to close this as there is no further action on this simple case.  If you have a more realistic case in the future please (re)open an issue for it.

closed 2021-07-20 by maliberty