The-OpenROAD-Project/OpenROAD #808: There was always an Error 2 happened when installing openroad

JiaDYuan, 2021-07-01:
[ 44%] Built target lefzlib
make[2]: Nothing to be done for `src/OpenDB/src/def/CMakeFiles/def.dir/build'.
[ 44%] Built target zutil
[ 44%] Built target def
make: *** [all] Error 2

I am running on MacOS, I have installed all the dependencies. Then run sudo ./etc/Build.sh. However whatever I tried, there was always an Error 2 happened....

maliberty, 2021-07-01:
You need to include more of the log.  Likely there was an error further up

JiaDYuan, 2021-07-01:
[ 44%] Swig compile OpenRoad-py.i for python
cd /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /usr/local/Cellar/cmake/3.20.1/bin/cmake -E make_directory /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir
cd /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /usr/local/Cellar/cmake/3.20.1/bin/cmake -E touch /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp
cmake -E touch: failed to update "/Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp".
make[2]: *** [src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp] Error 1
make[1]: *** [src/CMakeFiles/openroad_swig_py_swig_compilation.dir/all] Error 2
make: *** [all] Error 2

maliberty, 2021-07-01:
do you have swig installed?  Did it give an error?

JiaDYuan, 2021-07-01:
jiadyuan@JiadongdeMacBook-Pro build % brew install swig
Updating Homebrew...
==> Auto-updated Homebrew!
Updated Homebrew from 89aeaffd9 to e20ec394b.
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
s4cmd
==> Updated Formulae
Updated 18 formulae.
==> Updated Casks
Updated 7 casks.

Warning: swig 4.0.2 is already installed and up-to-date.
To reinstall 4.0.2, run:
  brew reinstall swig

JiaDYuan, 2021-07-01:
-- OpenROAD version: 1
-- OpenROAD git sha: cf744a3734204ef12378150d429acf32af385ea4
-- System name: Darwin
-- Compiler: AppleClang 12.0.5.12050022
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/local/lib/libtcl8.6.dylib
-- TCL header: /usr/local/include/tcl.h
-- TCL readline library: /opt/local/lib/libtclreadline.dylib
-- TCL readline header: /opt/local/include
-- Found Boost: /usr/local/include (found version "1.68.0")  
-- spdlog: 1.8.5
-- STA version: 2.3.0
-- STA git sha: 35a3f1e4e3f148b30678f9455e64d220c6758462
-- System name: Darwin
-- Compiler: AppleClang 12.0.5.12050022
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- TCL library: /usr/local/lib/libtcl8.6.dylib
-- TCL header: /usr/local/include/tcl.h
-- SSTA: 0
-- STA executable: /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GUI is enabled
-- Found Boost: /usr/local/include (found suitable version "1.68.0", minimum required is "1.68.0")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done

maliberty, 2021-07-01:
The stamp file is just a timestamp of when swig ran.  If it is failing it is likely because swig is failing.  Would you attach a full build log?

JiaDYuan, 2021-07-01:
Swig compile OpenRoad-py.i for python
cd /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /usr/local/Cellar/cmake/3.20.1/bin/cmake -E make_directory /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir
cd /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /usr/local/Cellar/cmake/3.20.1/bin/cmake -E touch /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp
cd /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /usr/local/Cellar/cmake/3.20.1/bin/cmake -E env SWIG_LIB=/opt/local/share/swig/4.0.2 /opt/local/bin/swig -python -Werror -w317,325,378,401,402,467,472,503,509 -outdir /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src -c++ -module openroad_swig_py -interface _openroad_swig_py -o /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/build/src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON_wrap.cxx /Users/jiadyuan/Desktop/OpenROAD-flow-scripts/tools/OpenROAD/src/OpenRoad-py.i
:3: Error: Unable to find 'python.swg'
make[2]: *** [src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp] Error 1
make[2]: *** Deleting file `src/CMakeFiles/openroad_swig_py.dir/OpenRoad-pyPYTHON.stamp'
make[1]: *** [src/CMakeFiles/openroad_swig_py_swig_compilation.dir/all] Error 2
make: *** [all] Error 2

Unable to find 'python.swg', what is this file?

maliberty, 2021-07-01:
That is part of the swig installation so there is something wrong with your install.
https://github.com/swig/swig/blob/master/Lib/python/python.swg

closed 2021-07-01 by maliberty

JiaDYuan, 2021-07-01:
Solved the problem, if installing the openroad on MacOS, need more dependencies, as for swig, still need swig-python

brew install swig-python

maliberty, 2021-07-01:
If you find the right recipe and want to contribute it back as documentation that would be welcome.