The-OpenROAD-Project/OpenROAD #849: Apparent Rounding Error When Creating Floorplan Rows

donn, 2021-07-13:
I want to initialize exactly 35 rows. The command I'm using is as follows:

```
        initialize_floorplan\
            -die_area "0 0 451.35 95.2"\
            -core_area "0.0 0.0 451.35 95.2"\
            -site unithd
```

There has been a regression since 4d4d7205fd0292dbf3fae55fad9109b3f0bd5786 where it creates one less row.

This appears to be a rounding or off-by-one of some kind given that adding +0.1 creates the correct number of rows.

Additionally, when using my custom placer, which utilizes the entire grid, the last *column* experiences a similar issue, where all cells placed on the last column fail `verify_placement`. Again, incrementing the width by +0.1 works around this issue.

```
OpenROAD 1 4d4d7205fd0292dbf3fae55fad9109b3f0bd5786
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./sky130_fd_sc_hd.merged.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./sky130_fd_sc_hd.merged.lef at line 2.

[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 437 library cells
[INFO ODB-0226] Finished LEF file:  ./sky130_fd_sc_hd.merged.lef
[INFO IFP-0001] Added 35 rows of 981 sites.
```

After:

```
OpenROAD 1 4cfaab83011661be0aef64288c0d5a76315c443c
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./sky130_fd_sc_hd.merged.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./sky130_fd_sc_hd.merged.lef at line 2.

[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 437 library cells
[INFO ODB-0226] Finished LEF file:  ./sky130_fd_sc_hd.merged.lef
[INFO IFP-0001] Added 34 rows of 981 sites.
```

Reproducible: [RAM32_SIZE32.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/6811216/RAM32_SIZE32.zip)
Instructions: run `fp_init.tcl` with the desired version of OR.

maliberty, 2021-07-13:
The value is 95.2 and it is scaled to meters by distance_ui_sta and back to microns in InitFloorplan::metersToDbu.  However the scaling to meters is done with a float not a double which leads to rounding issues.

In gdb:

(gdb) p value
$43 = 95.200000000000003
(gdb) p (value * 1e-6f) * 1e6
$44 = 95.199999759643106
(gdb) p (value * 1e-6) * 1e6
$45 = 95.200000000000003

$44 is what we are getting while $45 is what using a double would produce.  In InitFloorplan::makeRows:

  int rows_y = core_dy / site_dy;

rounds down to 34 instead of 35 as a result.  I think sta should use a double scale factor here.  @jjcherry56

QuantamHD, 2021-07-30:
I believe the previous PR fixed this @donn can you confirm

donn, 2021-07-30:
It did. Thanks, all.

closed 2021-07-30 by donn