The-OpenROAD-Project/OpenROAD #853: Prevent TritonRoute from creating custom Vias

quentinw91, 2021-07-14:
Hi,
I have defined horizontal and vertical Vias in my tech file:
```
VIA via2_V DEFAULT
  LAYER via2 ;
    RECT -0.035 -0.035 0.035 0.035 ;
  LAYER metal2 ;
    RECT -0.035 -0.07 0.035 0.07 ;
  LAYER metal3 ;
    RECT -0.035 -0.07 0.035 0.07 ;
END via2_V

VIA via2_H DEFAULT
  LAYER via2 ;
    RECT -0.035 -0.035 0.035 0.035 ;
  LAYER metal2 ;
    RECT -0.07 -0.035 0.07 0.035 ;
  LAYER metal3 ;
    RECT -0.07 -0.035 0.07 0.035 ;
END via2_H
```

But TritonRoute gives a warning that it needs to create a new custom VIA:
```
...
[INFO DRT-0151] Reading Guide

#guides:     3417

[WARNING DRT-0160] Warning: metal2 does not have viaDef aligned with layer direction, generating new viaDef via2_FR.
[WARNING DRT-0160] Warning: metal3 does not have viaDef aligned with layer direction, generating new viaDef via3_FR.
[INFO DRT-0167] List of default vias:
  Layer via1
    default via: via1_7
  Layer via2
    default via: via2_FR
  Layer via3
    default via: via3_FR
```

It creates new VIAs, with a cross shape like the following:
[image]

This works fine in this case, but on another platform (i cannot release the files), it causes DRC errors.  Is it possible to prevent TritonRoute from creating this custom VIAs ?  TritonRoute was previously able to route without the cross-shaped vias.  Or has it become mandatory ?

testcase if you want to reproduce it: 
[detail_route_gcd_test_plat2_base_2021-07-14_10-20.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6815473/detail_route_gcd_test_plat2_base_2021-07-14_10-20.tar.gz)

maliberty, 2021-07-14:
This code hasn't changed in a long time so I'm surprised you have a new issue.  Do you know what the via used previously was for this layer?  Do you only have square via enclosures in your other platform?

There is an internal switch to disable via generation but there is no user control currently.

maliberty, 2021-07-14:
Also what sort of rule does the generated via violate?

quentinw91, 2021-07-14:
Well, I am working on a platform where the VIA enclosure needs to be increased if another VIA is stacked on top of the first one.  So TritonRoute created the new VIAs with the default enclosure. And did not take the VIAs that I defined (with increased enclosure) to avoid this issue.  I guess it could be solved by redefining the enclosure in the tech file, but this cross shape also prevents insertion of double vias...
Anyway, I managed to find the internal switch and set it up, similar to what is done for the GF14 platform:
```
  if (DBPROCESSNODE == "GF14_13M_3Mx_2Cx_4Kx_2Hx_2Gx_LB") {
    VIAINPIN_BOTTOMLAYERNUM = 2;
    VIAINPIN_TOPLAYERNUM = 2;
    USENONPREFTRACKS = false;
    BOTTOM_ROUTING_LAYER = 4;
    TOP_ROUTING_LAYER = 18;
    ENABLE_VIA_GEN = false;
  }

//my modif
  if (DBPROCESSNODE == "test_plat2_Node") {
    ENABLE_VIA_GEN = false;
    logger_->info(utl::DRT,
                    251,
                    "my modif: Disabled the via generation. ENABLE_VIA_GEN is {}",
                    ENABLE_VIA_GEN);
```
This solved the issue.  Thank you

closed 2021-07-14 by quentinw91

maliberty, 2021-07-14:
I'm glad open-source allows you to fix your issue :)

We could consider making via selection/generation smarter in the future though it would be difficult without a LEF to work with.

quentinw91, 2021-07-14:
Yep, this is really nice !
This enclosure rule is quite special.  It is on an older technology, and I have never seen anything like this on another technology (even in the same foundry).  So to be honest, I am not sure that modifying the tool for this is worth it...  And I am not even sure that the LEF format lets you define the enclosure of a VIA depending on the fact that it is stacked or not.
I think having the tool try to search between a set of possible viaDEFs when routing (especially when trying to find access point to the cells) would be a more interesting upgrade.  As it would then allow to have full double VIAs capability :)

rovinski, 2021-07-15:
Seems like `ENABLE_VIA_GEN` should just be settable via tcl. I could make a quick PR if desired.

quentinw91, 2021-07-15:
That would be nice if you don't mind !

rovinski, 2021-07-18:
@quentinw91 this is done, you can use `detailed_route -disable_via_gen` with 1a9a8b6

PR #868 (2021-07-18): [DRT] Add -disable_via_gen flag to detailed_route
Resolves #853