The-OpenROAD-Project/OpenROAD #866: Fedora build error:

zml, 2021-07-17:
I ran into the following error while trying to build (manually) on Fedora 34 (5.11.20-300.fc34.x86_64).:

```
In file included from /home/$USER/sources/OpenROAD/src/TritonRoute/src/gr/FlexGRCMap.h:37,
                 from /home/$USER/sources/OpenROAD/src/TritonRoute/src/gr/FlexGRCMap.cpp:29:
/home/$USER/sources/OpenROAD/src/TritonRoute/src/frBaseTypes.h:331:20: error: ‘segment’ in namespace ‘boost::geometry::model’ does not name a template type
  331 | typedef bg::model::segment<point_t> segment_t;

```
I also share here the `cmake ..` output :

```
-- OpenROAD version: 1
-- OpenROAD git sha: 1ad7ce2303b8c5a56b4cbb9ef95866e4ac41c683
-- System name: Linux
-- Compiler: GNU 11.1.1
-- Build type: RELEASE
-- Install prefix: /home/$USER/.local
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/include (found version "1.75.0")  
-- spdlog: 1.8.5
-- STA version: 2.3.0
-- STA git sha: be0d38e382bbc7d07bc633ae87d1459816252c3d
-- System name: Linux
-- Compiler: GNU 11.1.1
-- Build type: RELEASE
-- Build CXX_FLAGS: -O2 -DNDEBUG
-- Install prefix: /home/$USER/.local
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- SSTA: 0
-- STA executable: /home/$USER/sources/OpenROAD/src/sta/app/sta
-- GUI is not enabled
-- Found Boost: /usr/include (found suitable version "1.75.0", minimum required is "1.68.0")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/$USER/sources/OpenROAD/build
```
Thanks

zml, 2021-07-17:
I am sorry, I have just re-read in the dependency list : **boost 1.68 (1.75 will not compile)**.
I am closing the issue.

closed 2021-07-17 by zml

rovinski, 2021-07-18:
@maliberty can we fix up the CMake find_package to better enforce this? Currently we just have:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/a5a56584fed1523f384e2944ee8439256ab5e12d/src/CMakeLists.txt#L70

We could do this if 1.68 is the only version that works:
```
find_package(Boost 1.68 EXACT REQUIRED)
```
Or this if 1.68 - 1.74 work:
```
find_package(Boost 1.68...1.74 REQUIRED)
```

maliberty, 2021-07-18:
That would probably be better.  It's been on the todo list for a while to look at what changed in the newer boost versions.  I'll try to take a look soon.