The-OpenROAD-Project/OpenROAD #907: finish opendb logger conversion

maliberty, 2021-07-22:
There are still a fair number of calls to notice and other functions in dbLogger.h.  Please finish the conversion to the new logger.

dralabeing, 2021-08-25:
@maliberty Is this conversion to logger complete?

titan73, 2024-09-03:
What exactly this conversion consists in?
Btw, is it still valid?

titan73, 2024-09-03:
So whenever I get a message without a prefix like "[INFO ODB-0227]" that means it's not converted yet, right?

maliberty, 2024-09-03:
Yes

rovinski, 2024-09-05:
It means it could also be using `logger->report()` though

maliberty, 2024-09-05:
> It means it could also be using `logger->report()` though

That should be limited to true reports but there have been some abuses that should also be fixed.  At least it is better than printf as it goes through the logger.

titan73, 2024-10-02:
Found some printf also in def parsing code:

```
def/def/defiFPC.cpp:              "ERROR (DEFPARS-6030): Invalid direction specified with FPC "
def/def/defiVia.cpp:            "ERROR (DEFPARS-6180): The index number %d specified for the VIA "
def/def/defiNet.cpp:  errMsg << "ERROR (DEFPARS-6084): The index number " << index
def/def/defiNet.cpp:  errMsg << "ERROR (DEFPARS-6085): The index number " << index
def/def/defiNet.cpp:  errMsg << "ERROR (DEFPARS-6086): The index number " << index
def/def/defiNet.cpp:        "ERROR (DEFPARS-6080): An internal error has occurred. The index "
def/def/defiNet.cpp:        "ERROR (DEFPARS-6081): An internal error has occurred. The index "
def/def/defiNet.cpp:        "ERROR (DEFPARS-6082): An internal error has occurred. The index "
def/def/defiNet.cpp:    errMsg << "ERROR (DEFPARS-6083): The index number " << index;
def/def/defiRegion.cpp:  msg << "ERROR (DEFPARS-6130): The index number " << index
def/def/defiRegion.cpp:  msg << "ERROR (DEFPARS-6131): The index number " << index
def/def/defiNonDefault.cpp:  msg << "ERROR (DEFPARS-6090): The index number " << index
def/def/defiNonDefault.cpp:  msg << "ERROR (DEFPARS-6091): The index number " << index
def/def/defrReader.cpp:                "WARNING (DEFPARS-5001): DEF statement found in the def file "
def/def/defiScanchain.cpp:              "ERROR (DEFPARS-6150): The START statement in the SCANCHAINS has "
def/def/defiScanchain.cpp:              "ERROR (DEFPARS-6151): The STOP statment in the SCANCHAINS has "
def/def/defiTimingDisable.cpp:              "ERROR (DEFPARS-6170): The TimingDisable type is invalid. The "
def/def/defiRowTrack.cpp:  msg << "ERROR (DEFPARS-6140): The index number " << index
def/def/defiIOTiming.cpp:              "ERROR (DEFPARS-6060): Invalid value specified for IOTIMING "
def/def/defiIOTiming.cpp:              "ERROR (DEFPARS-6060): Invalid value specified for IOTIMING "
def/def/defiPinProp.cpp:  msg << "ERROR (DEFPARS-6120): The index number " << index
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): %s, file %s at line %s\nLast token was "
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): %s, file %s at line %s\nLast token was "
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): Def parser has encountered an error in file "
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): %s, file %s at line %s\nLast token was "
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): %s, file %s at line %s\nLast token was "
def/def/def_keywords.cpp:              "ERROR (DEFPARS-%d): Def parser has encountered an error in file "
def/def/def_keywords.cpp:            "ERROR (DEFPARS-%d): %s Error in file %s at line %s, on token "
def/def/def_keywords.cpp:            "INFO (DEFPARS-%d): %s See file %s at line %s.\n",
def/def/def_keywords.cpp:            "INFO (DEFPARS-%d): %s See file %s at line %s.\n",
def/def/def_keywords.cpp:            "INFO (DEFPARS-%d): %s See file %s at line %s\n",
def/def/def_keywords.cpp:            "WARNING(DEFPARS-8500): Unable to open the file defRWarning.log in "
def/def/def_keywords.cpp:                "INFO (DEFPARS-%d): %s See file %s at line %s\n",
def/def/def_keywords.cpp:            "WARNING (DEFPARS-8500): Unable to open the file defRWarning.log "
def/def/def_keywords.cpp:                "INFO (DEFPARS-%d): %s See file %s at line %s\n",
def/def/def_keywords.cpp:            "WARNING (DEFPARS-%d): %s See file %s at line %s.\n",
def/def/def_keywords.cpp:            "WARNING (DEFPARS-%d): %s See file %s at line %s.\n",
def/def/def_keywords.cpp:            "WARNING (DEFPARS-%d): %s See file %s at line %s\n",
def/def/def_keywords.cpp:            "WARNING (DEFPARS-7500): Unable to open the file defRWarning.log "
def/def/def_keywords.cpp:                "WARNING (DEFPARS-%d): %s See file %s at line %s\n",
def/def/def_keywords.cpp:                "WARNING (DEFPARS-%d): %s See file %s at line %s\n",
def/def/defiSlot.cpp:  msg << "ERROR (DEFPARS-6160): The index number " << index
def/def/defiAssertion.cpp:                "ERROR (DEFPARSE-6009): An invalid attribute type has "
def/def/defiAssertion.cpp:        "ERROR (DEFPARS-6201): Unable to process the DEF file. Both WIREDLOGIC "
def/def/defiAssertion.cpp:        "ERROR (DEFPARS-6201): Unable to process the DEF file. Both WIREDLOGIC "
def/def/defiPartition.cpp:              "ERROR (DEFPARS-6100): The value spefified for PARTITION SETUP "
def/def/defiPartition.cpp:              "ERROR (DEFPARS-6101): The value spefified for PARTITION HOLD is "
```

Should they be converted to?

maliberty, 2024-10-02:
I don't think those are printf.  The msg is passed to defiError which should allow us to redirect it to the logger in one central place.

titan73, 2024-10-11:
Ah indeed. I'll send a PR for that.

titan73, 2025-01-27:
I encountered an error message not going through the logger:
`Error: port direction not found.`
Which comes from:
`sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: port direction not found."),`

After a search I found other cases:
```
rsz/src/Resizer.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown transition name."),
rsz/src/Resizer.i:    Tcl_SetResult(interp,const_cast<char*>("Error: parasitics source."), TCL_STATIC);
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: transition not found."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown rise/fall edge."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown transition name."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: port direction not found."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown timing role."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown logic value."),
sta/tcl/StaTclTypes.i:    Tcl_SetResult(interp,const_cast<char*>("Error: unknown analysis type."),
```

How this is supposed to be converted to the logger?

maliberty, 2025-01-27:
The ones in sta/tcl/StaTclTypes.i should be filed against https://github.com/parallaxsw/OpenSTA.

For rsz you could just use `Logger* logger = ord::getLogger();` and then issue the error.

titan73, 2025-01-27:
Ok thanks.
I've got something to compile for rsz. Any idea how I can trigger these 2 messages?

maliberty, 2025-01-27:
A simple test
```
penroad> rsz::estimate_parasitics_cmd blah blah
Error: parasitics source.
while evaluating rsz::estimate_parasitics_cmd blah blah
```

titan73, 2025-01-28:
Ok thanks.
I did:
```
    Logger* logger = ord::getLogger();
    logger->error(utl::RSZ, 19, "Unknown parasitics source '{}'.", arg);
```
It works with the gui:
```
>>> rsz::estimate_parasitics_cmd blah blah
[ERROR RSZ-0019] Unknown parasitics source 'blah'.
```
although the gui does not catch the exception ([ERROR-GUI-0070] ...)
without the gui it crashes:
```
openroad> rsz::estimate_parasitics_cmd blah blah
[ERROR RSZ-0019] Unknown parasitics source 'blah'.
terminate called after throwing an instance of 'std::runtime_error'
  what():  RSZ-0019
Signal 6 received
Stack trace:
 0# 0x0000590055B1C93B in openroad
 1# 0x00007F6FEA042520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F6FEA4A2B9E in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x00007F6FEA4AE20C in /lib/x86_64-linux-gnu/libstdc++.so.6
 7# 0x00007F6FEA4AE277 in /lib/x86_64-linux-gnu/libstdc++.so.6
 8# 0x00007F6FEA4AE4D8 in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# void utl::Logger::error<char const*>(utl::ToolId, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const* const&) in openroad
10# 0x0000590055CE8948 in openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F6FEED10028 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# 0x0000590055B1BD90 in openroad
16# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# main in openroad
18# 0x00007F6FEA029D90 in /lib/x86_64-linux-gnu/libc.so.6
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in openroad
Aborted (core dumped)
```
It seems the exception is not catched.

maliberty, 2025-01-29:
I suppose that's why its this way.  The Exception.i handler doesn't wrap this code.  You could log it in a try-catch and then return the error to tcl which will stop execution anyways.

titan73, 2025-02-05:
Yes indeed. Thanks. Just sent https://github.com/The-OpenROAD-Project/OpenROAD/pull/6647.

closed 2025-03-06 by maliberty