The-OpenROAD-Project/OpenROAD #908: Feature request: support "HALO" annotation in DEFs

nmoroze, 2021-07-22:
I want to be able to initialize a floorplan based on a provided DEF file where a component has a "HALO" annotation, and noticed this currently isn't supported (`initialize_floorplan` outputs "error: HALO on component is unsupported"). Are there plans to add this feature eventually? It'd be helpful for my use case.

Thanks!

maliberty, 2021-07-22:
It would be helpful to have the exact message which makes it easier to see where in the code the problem is.  Even better is a small test case.

maliberty, 2021-07-22:
I see, it is one of our old pre-logger messages.  Are you in need of ROUTEHALO too?  Are you applying this to a macro?  If so that macro placer supports a halo option though it has to be specified outside of DEF.

nmoroze, 2021-07-23:
Thanks for the prompt reply! We don't need ROUTEHALO, and yes, we are applying it to a macro. Thanks for the note about the macro placer - I think that could be a helpful workaround, but for our case it would ultimately be ideal to read the DEF file annotation (I'm trying to prepare a floorplan that could in theory be fed into a variety of tools with minimal tool-specific configuration).

maliberty, 2021-07-23:
I agree that the macro placer should be able to pick up the halo from DEF as well.  We are in the middle of bringing up a new macro placer so I'll suggest this for the new one (@ravi-varadarajan).

ravi-varadarajan, 2021-07-28:
@nmoroze  To confirm your use model you are initializing a floorplan with   read_def -floorplan_initialize and in the COMPONENT section of the DEF there is  +HALO which is currently throwing an error.  You would like this to be to used as the halo specification for the component.  Also if there is a global halo specification as in the call _**macro_placement**_  command, would this be additive?

maliberty, 2021-07-28:
@ahmadelrouby please take care of the opendb part of this problem (taking the halo from def and storing it).

ravi-varadarajan, 2021-07-28:
If it is stored as a PORPERTY on the COMPONENT, we can use during macro placement

maliberty, 2021-07-28:
There is a dbInst::getHalo but oddly not setHalo.  There is a field to store the data so we just need to hook it up and make it functional.  There is no field for a soft halo but we could add a bit if needed.

nmoroze, 2021-07-29:
> @nmoroze To confirm your use model you are initializing a floorplan with read_def -floorplan_initialize and in the COMPONENT section of the DEF there is +HALO which is currently throwing an error. [...]

That's correct! I haven't thought about how this would interact with a global halo specification. Maybe the HALO in the DEF could take priority, since it's more specific? I won't be needing a `macro_placement` command for my use case though (all macros will be placed in the provided DEF), so I have no specific needs for how those two should interact.