The-OpenROAD-Project/OpenROAD #910: ispd18_sample fails with diffs

yurivict, 2021-07-23:
```
*** /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/TritonRoute/test/ispd18_sample.ok    Fri Jul 16 14:09:30 2021
--- /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/TritonRoute/test/results/ispd18_sample.log   Thu Jul 22 23:44:18 2021
***************
*** 78,81 ****
  [INFO DRT-0036] Via8 guide region query size = 0
  [INFO DRT-0036] Metal9 guide region query size = 0
  [INFO DRT-0179] init gr pin query ...
! No differences found.
--- 78,83 ----
  [INFO DRT-0036] Via8 guide region query size = 0
  [INFO DRT-0036] Metal9 guide region query size = 0
  [INFO DRT-0179] init gr pin query ...
! Differences found at line 61.
!       + ROUTED Metal2 ( 92200 73530 ) ( * 78850 )
!       + ROUTED Metal2 ( 92200 73530 ) ( * 77710 )
```

maliberty, 2021-11-28:
There is no plan for OR to support FreeBSD.  We welcome patches but can't spend the resources to track down issues.

closed 2021-11-28 by maliberty

yurivict, 2021-11-28:
I didn't ask to support FreeBSD. This issue is about a QA failure.

maliberty, 2021-11-28:
There is no such failure on other platforms.