The-OpenROAD-Project/OpenROAD #911: ext_pattern, gcd, 45_gcd diffs

yurivict, 2021-07-23:
```
*** /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/ext_pattern.ok	Fri Jul 16 14:09:30 2021
--- /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/results/ext_pattern.log	Thu Jul 22 23:42:45 2021
***************
*** 11,17 ****
  [INFO RCX-0029] Defined extraction corner X
  [INFO RCX-0008] extracting parasitics of blk ...
  [INFO RCX-0435] Reading extraction model file ext_pattern.rules ...
! [INFO RCX-0436] RC segment generation blk (max_merge_res 0.0) ...
  [INFO RCX-0040] Final 70552 rc segments
  [INFO RCX-0439] Coupling Cap extraction blk ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
--- 11,17 ----
  [INFO RCX-0029] Defined extraction corner X
  [INFO RCX-0008] extracting parasitics of blk ...
  [INFO RCX-0435] Reading extraction model file ext_pattern.rules ...
! [INFO RCX-0436] RC segment generation blk (max_merge_res 0) ...
  [INFO RCX-0040] Final 70552 rc segments
  [INFO RCX-0439] Coupling Cap extraction blk ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
*** /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/gcd.ok	Fri Jul 16 14:09:30 2021
--- /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/results/gcd.log	Thu Jul 22 23:43:00 2021
***************
*** 17,23 ****
  [INFO RCX-0008] extracting parasitics of gcd ...
  Notice 0: Split top of 102 T shapes.
  [INFO RCX-0435] Reading extraction model file ext_pattern.rules ...
! [INFO RCX-0436] RC segment generation gcd (max_merge_res 0.0) ...
  [INFO RCX-0040] Final 3221 rc segments
  [INFO RCX-0439] Coupling Cap extraction gcd ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
--- 17,23 ----
  [INFO RCX-0008] extracting parasitics of gcd ...
  Notice 0: Split top of 102 T shapes.
  [INFO RCX-0435] Reading extraction model file ext_pattern.rules ...
! [INFO RCX-0436] RC segment generation gcd (max_merge_res 0) ...
  [INFO RCX-0040] Final 3221 rc segments
  [INFO RCX-0439] Coupling Cap extraction gcd ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
*** /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/45_gcd.ok	Fri Jul 16 14:09:30 2021
--- /disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0/src/rcx/test/results/45_gcd.log	Thu Jul 22 23:43:10 2021
***************
*** 15,21 ****
  [INFO RCX-0008] extracting parasitics of gcd ...
  Notice 0: Split top of 123 T shapes.
  [INFO RCX-0435] Reading extraction model file 45_patterns.rules ...
! [INFO RCX-0436] RC segment generation gcd (max_merge_res 0.0) ...
  [INFO RCX-0040] Final 2656 rc segments
  [INFO RCX-0439] Coupling Cap extraction gcd ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
--- 15,21 ----
  [INFO RCX-0008] extracting parasitics of gcd ...
  Notice 0: Split top of 123 T shapes.
  [INFO RCX-0435] Reading extraction model file 45_patterns.rules ...
! [INFO RCX-0436] RC segment generation gcd (max_merge_res 0) ...
  [INFO RCX-0040] Final 2656 rc segments
  [INFO RCX-0439] Coupling Cap extraction gcd ...
  [INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
```

maliberty, 2021-07-23:
Similar issue in https://github.com/The-OpenROAD-Project/OpenROAD/pull/806

mooredan, 2021-07-23:
That pull request fixes this by adding formatting to the RCX-0436 message for the _mergeResBound double variable in src/rcx/src/netRC.cpp

mooredan, 2021-07-25:
@yurivict Please pull from master and verify if this issue is resolved.

yurivict, 2021-07-25:
Thanks.

I will verify with the next release.

closed 2021-10-18 by dralabeing