The-OpenROAD-Project/OpenROAD #921: [ERROR DRT-0073] no ap for Instance/pin in a design with just 8% utilization

zml, 2021-07-26:
As mentioned in the Gitter discussion, I have been constantly getting the **[ERROR DRT-0073] no ap for (...)** when running the `detailed_route` step. 

In an effort to find out the root cause for this issue I have reduced the design to a [ Minimum Working Example - mwe.tar.gz](https://drive.google.com/file/d/1hHTpQJTrkNwJY7ddwkiOwFEsqDvCgXsX/view?usp=sharing.).

This design example has   8% utilization, so this is should not be related with excessive routing congestion.. 
In order to easily replicate the issue, please proceed as follows:

[Download **mwe.tar.gz**](https://drive.google.com/file/d/1hHTpQJTrkNwJY7ddwkiOwFEsqDvCgXsX/view?usp=sharing.) to **OpenROAD-flow-scripts/** directory, untar it to **OpenROAD-flow-scripts/mwe/** and then run:

```
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) --mount type=bind,source=$(pwd)/flow/platforms,target=/OpenROAD-flow-scripts/flow/platforms --mount type=bind,source=$(pwd)/mwe,target=/mwe  openroad/flow-scripts
```
or, alternatively,  run
```
source mwe/run.sh
```
then inside the running docker container, run the following:
```
source setup_env.sh 
openroad /mwe/cmd.tcl 
```
Please let me know if there is something else I can do to aid in the debug.
Thank you in advance!

Stephanommg, 2021-07-26:
The problem is with STDLIB_3bdec cell, pin yz[0]. You are creating a port for each rect, when you could create a single port for all of them (at least for those that are in touch with each other). If you create a port for all the pin shapes it should work. Still, the macro definition is not wrong and I think TR should handle this situation, but if you want a quick solution, you can do as I suggested.

[image]

maliberty, 2021-07-26:
Putting touch shapes in different ports is wrong, see the LEF manual:
   Each PORT of the same PIN is considered weakly connected to the other PORTs

If they are touching shapes they are not weakly connected.  We might be able to route anyways but it is not correct modeling.

zml, 2021-07-27:
Thanks for such a quick feedback.
I confirm that modifying the LEF in this way solves the issue and I am able to get a fully routed DEF on this design sample! 

Still, I am now trying to do the same on the original design and I get this error one just one cell. 
I am trying to check if this is still a LEF issue with the particular std cell.

Do you have any  hints on how to debug this kind of errors ? 
Is there anyway to output a partial routed DEF file for debug whenever this kind of errors arise ?

zml, 2021-07-27:
I have been able to avoid this error by slightly changing the pin location of STDCELL. 
Thanks very much for the great support.

zml, 2021-07-28:
I am sorry to re-open the issue, but could you please help me debug a similar error, on a different STD cell ? 
This time I do not have _weakly connected_ PORTs, so I guess it should be related with a different issue here.
The error I get is  `[ERROR DRT-0073] no ap for I150_16581/s`, which corresponds to the **s** port of a SR-latch.

The files are at: [ Minimum Working Example 2 - mwe2.tar.gz](https://drive.google.com/file/d/1oxzPsNqyFzfY3CsswZtvDUm3s2rDANNz/view?usp=sharing)

I really want to understand how to debug this issues so I can avoid them in future designs.
Thank you once again for the great feedback.

Stephanommg, 2021-07-28:
Hi zml, no worries.

For debugging, you can used the flag -gui when you call openroad, but you also need to add, in you cmd.tcl, before detailed_route command, the following command:

detailed_route_debug -pa

You can also use the flag -pin (for this command) to track a specific pin. Use the following format for pin name: <instName>:<pinName>. Example:

detailed_route_debug -pa -pin inst0:A

However, I dont know if this will help much. My debugging required me to look at the code. But at least you can understand the context and narrow the error.

I will take a look on the new error right now.

Stephanommg, 2021-07-28:
The problem is that you have an obstacle over a small pin and TR is trying to create access points accessible only by vias.

[image]

I will further investigate why it is not allowing planar access.

maliberty, 2021-07-28:
@Stephanommg this sort of situation is what nearby grid APs were created for.  There is a similar situation in skywater.  I'm not sure if they are possible for this shape or not.  In general this isn't a good cell design practice.

Stephanommg, 2021-07-28:
This is the full picture with nearby aps. It seems that any via that will be used will be still close to the obstacle.
[image]

@zml By default TR doesnt allow planar access to access points in M1.

Stephanommg, 2021-07-28:
@zml you can set VIA_ACCESS_LAYERNUM in src/global.cpp to something less then 2 to allow planar access to those access points, but do this at you own risk since this is non-default in TR.

zml, 2021-07-28:
> @zml By default TR doesnt allow planar access to access points in M1.

Just to make sure I am getting this right.
Does that  mean that TR will always try to connect with M2 + Via directly and would not try to create a vertical M1 path  ?
Because if simply increase M1 rectangle height of the  **PIN s**   as follows :
```
#        RECT 2.11 3.63 2.245 3.87 ;
          RECT 2.11 3.63 2.245 5.7 ;
```
The error no longer occurs. 
Thanks

maliberty, 2021-07-28:
Right, in general routing on m1 may make other pins inaccessible.  To avoid those complications we just assume via access and check for compatibility of via access.  More work could be put into this area but that's the current strategy and it tends to work ok when pins are not blocked like this one.

Making the pin bigger likely means we find a way to drop a via on it and the problem is gone.

zml, 2021-07-28:
I get it. Many thanks.

closed 2021-07-28 by zml