The-OpenROAD-Project/OpenROAD #964: Dependency Issue

praharshapm, 2021-08-04:
I am trying to install the dependencies using etc/DependencyInstaller.sh, but I am getting error that:
E: The repository 'http://ppa.launchpad.net/saltmakrell/ppa/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

I am attaching screenshot as well. Am I missing any step?
[image]

vvbandeira, 2021-08-04:
@praharshapm this looks unrelated to OpenROAD. Does the error also occurs if you run `sudo apt update`?

praharshapm, 2021-08-04:
Oh, you are right. I am also getting same error for `sudo apt update`. Looks like there is some other issue. Thanks

closed 2021-08-04 by vvbandeira