The-OpenROAD-Project/OpenROAD #965: Routing "Killed"

donn, 2021-08-05:
https://github.com/The-OpenROAD-Project/OpenLane/issues/497

[05-08_13-27_or_groute_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/6938507/05-08_13-27_or_groute_packaged.tar.gz)

maliberty, 2021-08-09:
I ran it with the head of OR (cb60cd1859458b0089c7d9bdb483bc95cf1c9ce3) without error.

maliberty, 2021-08-09:
Btw, if it dying it will be in repair_antennas not global_route.

maliberty, 2021-08-10:
@donn would you test the latest OR in OL and see if this is resolved?

donn, 2021-08-12:
It appears that there is a regression of some kind (I would wager another metric’s output has changed,) so I will need to take a look at that first. Apologies for the delay.

maliberty, 2021-08-24:
@donn should I close this?