The-OpenROAD-Project/OpenROAD #1008: [ERROR PDN-9999] Unexpected error: can't read "stripe_locs(met4,GROUND)": no such element in array

donn, 2021-08-19:
```log
OpenROAD 1 7ff717102e1699417600b0d0df4025348124d824
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[WARNING ODB-0220] WARNING (LEFPARS-2036): SOURCE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file ./tmp/merged_unpadded.lef at line 973.

[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: Users/donn/efabless/openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def
[INFO ODB-0128] Design: spm
[INFO ODB-0130]     Created 36 pins.
[INFO ODB-0131]     Created 433 components and 2537 component-terminals.
[INFO ODB-0133]     Created 312 nets and 997 connections.
[INFO ODB-0134] Finished DEF file: Users/donn/efabless/openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: pdk/sky130A/libs.tech/openlane/common_pdn.tcl
[INFO PDN-0008] Design name is spm.
[INFO PDN-0009] Reading technology data.
[INFO PDN-0011] ****** INFO ******
Type: stdcell, grid
    Stdcell Rails
      Layer: met1  -  width: 0.480  pitch: 2.720  offset: 0.000 
    Straps
      Layer: met4  -  width: 1.600  pitch: 153.600  offset: 16.320 
      Layer: met5  -  width: 1.600  pitch: 153.180  offset: 16.650 
    Connect: {met4 met5} {met1 met4}
Type: macro, CORE_macro_grid_1
    Macro orientation: R0 R180 MX MY R90 R270 MXR90 MYR90
    Straps
    Connect: {met4_PIN_ver met5}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - grid.
can't read "stripe_locs(met4,POWER)": no such element in array
```

[openlane_test_or_pdn_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7012651/openlane_test_or_pdn_packaged.tar.gz)

donn, 2021-08-31:
We're blocked on this, btw.

tspyrou, 2021-09-01:
@Colin-Holehouse or @ibrahimkhairy can one of you pick this up urgently? OpenLane can't uprev to the latest openroad until this is fixed.

tspyrou, 2021-09-01:
@ibrahimkhairy Colin has a higher priority issue, can you try to debug this?

ibrahimkhairy, 2021-09-01:
ok, I'll have a look

ibrahimkhairy, 2021-09-03:
Hi Donn, 
 
 I just checked with Colin today and the test case needs 2 changes to pass: 
1) add voltage_domans definitions in pdngen config:
`set pdngen::voltage_domains {
   CORE {
    primary_power VPWR primary_ground VGND
  }
 }
`
 
 2) Change the variables FP_PDN_HPITCH and FP_PDN_VPITCH to a value that fits in the design; it was previously set to 153.18 which is larger than design dimentions
 
 I've attached a modified working test cases with changes to the following scripts:
 "run" ,  "run_ol", "common_pdn.tcl"
 
 Also, we'll have a PR later today to allow ICeWall to detect these cases automatically in future; it'll have the following updates: 
 1) pdngen will initialize voltage_domains if not set in the config
 2) will return a detailed error message when strap pitch exceeds design width or height

[updated_openlane_test_or_pdn_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7106063/updated_openlane_test_or_pdn_packaged.tar.gz)

donn, 2021-09-06:
```
add voltage_domans definitions in pdngen config:
set pdngen::voltage_domains { CORE { primary_power VPWR primary_ground VGND } } 
```

This is suddenly required... why?

donn, 2021-09-06:
merged the voltage domains thing into open_pdks and openroad. thanks for the help

closed 2021-09-06 by donn