The-OpenROAD-Project/OpenROAD #1010: Review and update MPL documentation

vvbandeira, 2021-08-19:
@mgwoo could you check if the commands and options are up to date? Also @abk-openroad asked if in the first paragraph of the [README.md](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/mpl/README.md#macro-placement) where it says it requires `global_placement` before running macro placement is for MS-RePlAce?  If so, then please say so explicitly.

mgwoo, 2021-08-20:
Hello Vitor,
The global_placement should be done before running macro_placement -- the MacroPlacer takes inputs from the placed solution from MS-RePl. (The ORFS always runs MS-RePl before macro_placement command -- stage 2-3 and 2-4)

jjcherry56, 2021-08-24:
I reviewed it (since I wrote it)

closed 2021-08-24 by jjcherry56