The-OpenROAD-Project/OpenROAD #1011: Review and update PPL documentation

vvbandeira, 2021-08-19:
@eder-matheus could you review the commands and arguments for ppl on the [README.md](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/ppl/README.md)?

Also the Pin Placement for the Top Layer features is missing. Maybe @arlpetergadfort could help with motivation/use case for these.

eder-matheus, 2021-08-19:
Everything looks correct to me. I think the Pin Placement for the Top Layer can be described as a combination of `define_pin_shape_pattern` and `set_io_pin_constraint` command. When you create a placement grid with the first command, you can restrict a set of pins to this grid with the second command. But I agree that a motivation for this feature may help us to define a better section for this functionality.

arlpetergadfort, 2021-08-20:
@eder-matheus @vvbandeira  I'll see if we can put something together to help motive that.

arlpetergadfort, 2021-08-20:
@eder-matheus  @vvbandeira Here is what I from my team member (the short answer is the command enables face-to-face direct-bonding IOs):
Studies have shown that 3D integration offers power and performance benefit. There is also a newly emerging opportunity for 3DIC application in the area of design obfuscation. Micro bumps and hybrid bonding are two of the more advanced packaging technologies that are used for 3D integration via face-to-face stacking [1](https://semiengineering.com/bumps-vs-hybrid-bonding-for-advanced-packaging/) The hybrid bonding technology provides even denser interconnect for the 3D stack [2](https://www.3dincites.com/2018/04/hybrid-bonding-from-concept-to-commercialization/) [3](https://en.wikipedia.org/wiki/Direct_bonding) [4](https://en.wikipedia.org/wiki/Direct_bonding). The implementation of face-to-face stacking involves placement of the ports/pins of the individual chips in the die area during physical design steps.

PR #1088 (2021-09-16): PPL add doc note on f2f usage
Fixes #1011
closed 2021-09-16 by vvbandeira