The-OpenROAD-Project/OpenROAD #1014: DPL documentation questions

vvbandeira, 2021-08-19:
On the DPL documentation [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/dpl/README.md), for the command `set_placement_padding` the option `-masters` is not mentioned on the description.

The command `set_power_net` is mentioned on the text, but is missing from the "block code" that shows the synopsis of all commands. Is this still a valid command? If so, what is its options?

In the description of `-max_displacement` we say that the default values are in "sites", but the next sentence says the unit is "microns". Also, it seems as if (500,100) originally contemplated placement sites in x, and cell row pitches in y. If these are now in microns, should the x and y max displacements be more similar to each other?

For the `filler_placement` command is `FILLER_` the default prefix? If so, maybe first saying (explicitly) the default value and then saying how to overwrite might more clear? Example:

> The default prefix is `FILLER_`, to specify a different naming prefix use -prefix <new prefix>. Wildcard matching is supported, so `FILL*` will match, e.g., `FILLCELL_X1 FILLCELL_X16 FILLCELL_X2 FILLCELL_X32 FILLCELL_X4 FILLCELL_X8`. 

Also, while addressing these questions it might be an opportunity to review the documentation to see if all commands and options are current.

maliberty, 2021-08-19:
@tspyrou please discuss this with Cherry

vijayank88, 2023-06-14:
@vvbandeira 
Is this still relevant?

vvbandeira, 2023-06-20:
@vijayank88 I have updated the issue with what is currently relevant.