The-OpenROAD-Project/OpenROAD #1017: Can't compile in MacOS when GUI is enabled

eder-matheus, 2021-08-20:
Logfile with the errors attached.
[compile_error.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/7024189/compile_error.log)

This compile error wasn't occurring before Pull Request https://github.com/The-OpenROAD-Project/OpenROAD/pull/1006

arlpetergadfort, 2021-08-20:
Not sure why it would break with that. Is it possible that the header for unordered_map is missing and that is the issue?
Would you be able to try adding `#include <unordered_map>` to line 41 in mainWindow.h?

I see another issue where the `keyPressEvent` is incorrectly a public, so that should be moved as well.

eder-matheus, 2021-08-23:
Thanks, @arlpetergadfort. Adding the `#include <unordered_map>` fixed the compile error for me.

arlpetergadfort, 2021-08-23:
Great, I'll get the PR approved so this is solved.

arlpetergadfort, 2021-08-23:
@eder-matheus I did not see your comment about the segfault, so I'll reopen. Is that intermittent? or when loading a specific design? Does it happen on load or when you select a path (looks like load but please confirm)?

eder-matheus, 2021-08-23:
@arlpetergadfort It happens even if I don't load any design. If I run just ´openroad -gui´, it crashes sometimes. I've just tested it, and I had to run ´openroad -gui´ five consecutive times to get the segfault.

arlpetergadfort, 2021-08-23:
okay, I'll take a look, there is probably a problem in the staGui code, which just wasn't an issue before switching it over to a dockwidget.

arlpetergadfort, 2021-08-23:
It's possible there an uninitialized variable. 
Can you try to replace the following in staGui.h (line 199)
`  TimingPathDetailModel() {}` ->   TimingPathDetailModel() { path_ = nullptr; }`
Does that help?

maliberty, 2021-08-23:
@eder-matheus can you valgrind startup on a mac (or maybe even on linux)?

eder-matheus, 2021-08-23:
@arlpetergadfort This replacement seems to work. I've run `openroad -gui` multiple times and I didn't face any segfault.

@maliberty Sure, I can (and will) run valgrind on centos.

arlpetergadfort, 2021-08-23:
the `path_` variable is definitely not initialized (based on the code), so that should probably get fixed as well.
Here is the branch with that change: https://github.com/The-OpenROAD-Project-private/OpenROAD/tree/gui-fix-macos-segfault

arlpetergadfort, 2021-08-23:
> This is flagged in Coverity
> [image]
> 
> @arlpetergadfort do you have access to look at the gui issues in Coverity?

@maliberty no I do not, if you want to give me access, I'd be happy to take a look.

@eder-matheus Could you check the branch I listed above and see if that solves it as well (same change, just modified where the initialization happens)?

eder-matheus, 2021-08-23:
@arlpetergadfort I've tested your branch and it isn't crashing anymore. I've also run Valgrind and it looks good too.

arlpetergadfort, 2021-08-23:
> @arlpetergadfort I've tested your branch and it isn't crashing anymore. I've also run Valgrind and it looks good too.

Great, in that case I'll merge that in.

closed 2021-08-23 by arlpetergadfort