The-OpenROAD-Project/OpenROAD #1027: ODB swig python API getPoint

osamahammad21, 2021-08-25:
odb has removed getPoint_ext API and used getPoint for both (with/without ext). getPoint with 2 arguments is wrapped while getPoint with 3 arguments isn't. So when I call getPoint expecting 3 returned values, I only get 2.

maliberty, 2021-08-25:
Are you planning to fix this or @ahmadelrouby ?

NairiBaveyann, 2024-06-05:
@maliberty @osamahammad21 @tspyrou
Hi, I wanted to work on this issue, but I have a doubt that the issue is obsolete because I see that OpenDB has been changed a lot since this issue has been created and the Python unit tests are outdated (paths, modules are incorrect and so on).
Currently, I see that there is no getPoint_ext API available, if it's still worth adding I can work on it.