The-OpenROAD-Project/OpenROAD #1028: ODB Python API new_int helper removed

osamahammad21, 2021-08-25:
new_int was a helper function to represent an int object that can be passed by reference to odb's APIs. It was removed recently. We don't need to add it again, but we need to handle functions returning by reference.

20Mhz, 2021-09-03:
Hi!, I got stuck trying to align openlane  to the lastest openroad ... do you have a suggestion for following code?
```
origin, count, step = odb.new_int(0), odb.new_int(0), odb.new_int(0)
block_top.findTrackGrid(H_LAYER).getGridPatternY(0, origin, count, step)
origin, count, step = odb.get_int(origin),  odb.get_int(count), odb.get_int(step)

```

20Mhz, 2021-09-09:
Took another look, it seems there was a pretty easy fix. Bests.
```
#origin, count, step = odb.new_int(0), odb.new_int(0), odb.new_int(0)
#block_top.findTrackGrid(H_LAYER).getGridPatternY(0, origin, count, step)
#origin, count, step = odb.get_int(origin),  odb.get_int(count), odb.get_int(step)
#block_top.findTrackGrid(H_LAYER).getGridPatternY(0, origin, count, step)
origin, count, step = block_top.findTrackGrid(H_LAYER).getGridPatternY(0)
```

closed 2021-11-28 by maliberty