The-OpenROAD-Project/OpenROAD #1049: Unable to build OpenROAD on CentOS7

legendmaker-wty, 2021-09-01:
Hi
I am trying to install OpenROAD on CentOS 7. I referred to the readme file on github and ran`./etc/DependencyInstaller.sh -dev`  for the dependencies. When I tried to compile the project with `cmake3 ..` , an error occurred:

> -- The CXX compiler identification is GNU 4.8.5
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ - works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- OpenROAD version: 1
> -- OpenROAD git sha: 527cd5d8e82308baa1a46265c1338e6c1757858f
> -- System name: Linux
> -- Compiler: GNU 4.8.5
> -- Build type: RELEASE
> -- Install prefix: /usr/local
> -- TCL library: /usr/lib64/libtcl.so
> -- TCL header: /usr/include/tcl.h
> -- Found SWIG: /usr/bin/swig (found suitable version "3.0.12", minimum required is "3.0")
> -- Found Boost: /usr/include (found version "1.53.0")
> -- Found Python3: /usr/lib64/libpython3.6m.so (found version "3.6.8") found components: Development
> -- Found ZLIB: /usr/lib64/libz.so (found version "1.2.7")
> CMake Error at src/CMakeLists.txt:107 (find_package):
> By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
> asked CMake to find a package configuration file provided by "spdlog", but
> CMake did not find one.
> Could not find a package configuration file provided by "spdlog" with any
> of the following names:
> spdlogConfig.cmake
> spdlog-config.cmake
> Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
> "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
> provides a separate development package or SDK, be sure it has been
> installed.
> -- Configuring incomplete, errors occurred!
> See also "/home/weitianyu/OpenROAD/build/CMakeFiles/CMakeOutput.log".

I tried to fix it with `sudo yum install spdlog` but it didn't help. 
Thanks.

vvbandeira, 2021-09-06:
@legendmaker-wty were you able to build?

legendmaker-wty, 2021-09-06:
No, the problem remains but I changed a platform and it worked.
Thanks for asking.

mithro, 2021-09-06:
BTW @vvbandeira - Shouldn't cmake complain that your compiler version is too old?

vvbandeira, 2021-09-06:
@legendmaker-wty thanks for letting us know, will close the issue.

@mithro I think this is a good idea. I will add the check, thanks.

closed 2021-09-06 by vvbandeira

legendmaker-wty, 2021-09-06:
I'd like to add some followup. I tried to build OpenROAD again on my CentOS
7. I referred to
https://stackoverflow.com/questions/53310625/how-to-install-gcc8-using-devtoolset-8-gcc
and ran the following 4 lines to enable the gcc-8 and for this time the
build succeeded. Thank you very much for your help!

yum install centos-release-scl
yum install devtoolset-8-gcc devtoolset-8-gcc-c++
scl enable devtoolset-8 -- bash

source /opt/rh/devtoolset-8/enable