The-OpenROAD-Project/OpenROAD #1052: OpenROAD version number?

aolofsson, 2021-09-02:
Is there currently a way to automatically check version compliance?
Currently a hash key is returned, but it's not clear how this key can be used to check the actual version?
￼
$ openroad -version
1 af9a0f9faafb7e61ae18e9496169c3527312b82a

As a comp, this is what verilator/yosys returns:
$ verilator --version
Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1

$ yosys -version
Yosys 0.9+3672 (git sha1 014c7e26, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)

Any thoughts about implementing a python like versioning methodology?

eg:
matplotlib >= 3.3
numpy >= 1.19

rovinski, 2021-09-02:
Technically the `1` is the version number, as listed here: https://github.com/The-OpenROAD-Project/OpenROAD/blob/4d956498152d921d73c06ca161f90356d7bba3a5/CMakeLists.txt#L48-L50

I don't think there has been very much focus on versioning, because we don't publish binaries. The git hash is technically suitable to identify the version, but It's not very human-friendly unfortunately. I have noticed that yosys has a bot that will automatically bump the version number whenever the master branch is updated. It would be nice if we had a similar strategy.

aolofsson, 2021-09-03:
Thanks! A couple of digits of precision after the 1 would be appreciated and a bot is a great idea.

maliberty, 2021-09-03:
One challenge of having version numbers is that people will report issues against older versions rather than against the head.  We are not staffed to support N releases so we would have to be clear that version numbers don't represent support.  Support is only at the head of master.

maliberty, 2021-09-03:
It would be simple to include the date associated with the commit (ie Thu Sep 2 14:08:55 2021 -0700).  Dates form a natural ordering that is easy to compare.

aolofsson, 2021-09-04:
In terms of support, I really people read and internalize the BSD license:-) "THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED.

In terms of using the date, I agree that it would work. It's not that common, but it would work (for me at least).

```python
import datetime
old = datetime.datetime(2021, 1, 9)
new = datetime.datetime(2021, 9, 1)
print("new is newer than old: ", old < new)

```

mithro, 2021-09-06:
Use `git describe` to get a usable version number from a git repository. Date's in Git don't work as they are not guaranteed to be committed in any order.

maliberty, 2021-09-06:
If I try this on the current app I get:

closed 2021-09-07 by maliberty

aolofsson, 2021-09-07:
My recommendation/preference would still be something  more systematic like the following.

Verilator 4.211 devel rev v4.210-26-gcf39331d

https://semver.org/

matplotlib >= 3.3
numpy >= 1.19
pytest >= 6.2.4
pandas >= 1.2.3
Jinja2 >= 2.11.3
Sphinx >= 3.5.4