The-OpenROAD-Project/OpenROAD #1053: Triton Route Won't build due boost issue

20Mhz, 2021-09-02:
I'm trying to build 1415572a73d43dfc91ba4501170fe6a245729e8. Using boost 1.76.0.  I'm getting:
```
[ 88%] Building CXX object src/TritonRoute/CMakeFiles/TritonRouteLib.dir/src/gr/FlexGR_rq.cpp.o
cd /Users/ronaldv/Projects/repositories/OpenROAD/build/src/TritonRoute && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DBOOST_ALL_NO_LIB -DFMT_LOCALE -DFMT_SHARED -DOPENROAD_BUILD -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -I/Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/include -I/usr/local/Cellar/tcl-tk/8.6.11_1/include -I/Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/src -I/Users/ronaldv/Projects/repositories/OpenROAD/include -I/Users/ronaldv/Projects/repositories/OpenROAD/src/OpenSTA/include -I/Users/ronaldv/Projects/repositories/OpenROAD/src/OpenDB/include -I/Users/ronaldv/Projects/repositories/OpenROAD/src/OpenDB/src/def/def -I/Users/ronaldv/Projects/repositories/OpenROAD/src/OpenDB/src/lef/lef -I/Users/ronaldv/Projects/repositories/OpenROAD/src/gui/include -I/Users/ronaldv/Projects/repositories/OpenROAD/src/OpenDB/include/opendb -I/Users/ronaldv/Projects/repositories/OpenROAD/src/flute3 -isystem /usr/local/include -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/opt/qt/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -Wall -pedantic -Wcast-qual -Wredundant-decls -Wformat-security -fPIC -Xclang -fopenmp -std=c++17 -MD -MT src/TritonRoute/CMakeFiles/TritonRouteLib.dir/src/gr/FlexGR_rq.cpp.o -MF CMakeFiles/TritonRouteLib.dir/src/gr/FlexGR_rq.cpp.o.d -o CMakeFiles/TritonRouteLib.dir/src/gr/FlexGR_rq.cpp.o -c /Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/src/gr/FlexGR_rq.cpp
In file included from /Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/src/gr/FlexGR_rq.cpp:29:
In file included from /Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/src/gr/FlexGR.h:35:
In file included from /Users/ronaldv/Projects/repositories/OpenROAD/src/TritonRoute/src/frRTree.h:34:
In file included from /usr/local/include/boost/geometry/algorithms/equals.hpp:26:
In file included from /usr/local/include/boost/geometry/algorithms/detail/equals/implementation.hpp:46:
In file included from /usr/local/include/boost/geometry/algorithms/relate.hpp:15:
In file included from /usr/local/include/boost/geometry/algorithms/detail/relate/implementation.hpp:26:
In file included from /usr/local/include/boost/geometry/algorithms/detail/relate/multi_point_geometry.hpp:33:
In file included from /usr/local/include/boost/geometry/index/rtree.hpp:64:
In file included from /usr/local/include/boost/geometry/index/detail/rtree/node/node.hpp:44:
In file included from /usr/local/include/boost/geometry/index/detail/algorithms/bounds.hpp:18:
In file included from /usr/local/include/boost/geometry/index/detail/bounded_view.hpp:23:
/usr/local/include/boost/geometry/strategies/index/services.hpp:31:5: error: static_assert failed due to requirement 'boost::geometry::detail::static_assert_check<false, fr::frBox, boost::geometry::cartesian_tag>::value' "Not implemented for this coordinate system."
    BOOST_GEOMETRY_STATIC_ASSERT_FALSE(
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

maliberty, 2021-09-02:
I believe this was fixed later on - can you try the head of master and see if that works for you?

20Mhz, 2021-09-02:
Hi Matt, 
That is correct, I've just succeeded building master. I'm actually trying to align to OpenLane which is using the checkin above. Can point me to the closest  stable or preferable to the checkin that address this issue?
Thanks,
Ronald

maliberty, 2021-09-02:
Are you using https://github.com/The-OpenROAD-Project/OpenLane or https://github.com/efabless/OpenLane.  The former is more current and should have a more recent OR version.  The later is frozen for the MPW2 shuttle.

20Mhz, 2021-09-02:
https://github.com/efabless/openlane.git

maliberty, 2021-09-02:
That version is frozen and will not update OR versions.  I suggest you use https://github.com/The-OpenROAD-Project/OpenLane

20Mhz, 2021-09-03:
I moved to main on that repo and I'm all set  for now. Thanks

closed 2021-09-03 by 20Mhz