The-OpenROAD-Project/OpenROAD #1058: PDN creates two power nets.

shuolai, 2021-09-03:
I was wondering if OpenROAD supports adding two power nets and one ground net as power rings? 
I tried several PDN command but it seems that I can only add a pair of power rings. (VDD and VSS)
Is there any method I can add another power ring VREG at the same time? 

The following image is what I am expecting.
[image]

arlpetergadfort, 2022-04-26:
@maliberty yes, I think so. With -secondary_net options.  It would be good if the OP could verify it works for them, but there are test cases that address this in pdn

vijayank88, 2023-06-14:
@shuolai 
Can you check latest pdn option?

gadfort, 2024-10-18:
@maliberty is this still a thing?

closed 2024-10-18 by maliberty