The-OpenROAD-Project/OpenROAD #1061: how to open gui for openroad

riya1407, 2021-09-04:
hello 
i am a student and working on openroad for a project .May I know the commands for opening gui of openroad or do i need to install anything else for the same ??

maliberty, 2021-09-04:
Just run openroad -gui

riya1407, 2021-09-06:
what to do if error occurs like the code was compiled with gui disabled . I
am attaching the screenshot with this email .
thanking you

maliberty, 2021-09-06:
You need to have Qt installed and rebuild the software.  The simplest is to run etc/DependencyInstaller.sh

riya1407, 2021-09-07:
i already have installed Qt .what changes are required in buil.sh for
rebuilding

vvbandeira, 2021-09-07:
@riya1407 could you run the command below and paste here the output from `output.txt`?

```
./etc/Build.sh -clean 2>&1 | tee output.txt
```

maliberty, 2021-09-07:
Note that you need to have Qt5 installed; Qt4 won't work.

When running cmake you should see:
-- GUI is enabled

riya1407, 2021-09-07:
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 1
-- OpenROAD git sha: 5df98f89ecfaa80be41b8ee2b0402594a93208c0
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.1", minimum
required is "3.0")
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake
(found version "1.71.0")
-- Found Python3: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version
"3.8") found components: Development
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- spdlog: 1.5.0
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- STA version: 2.3.0
-- STA git sha: 21ec025dab29c4badccfff276125b98e5ed47325
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- STA executable: /home/riya/OpenROAD/src/sta/app/sta
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GUI is not enabled
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake
(found suitable version "1.71.0", minimum required is "1.68.0")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/riya/OpenROAD/build
/usr/bin/cmake -S/home/riya/OpenROAD -B/home/riya/OpenROAD/build
--check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/riya/OpenROAD/build/CMakeFiles
/home/riya/OpenROAD/build/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all

the above pasted information i am getting after running the command given
by you

vvbandeira, 2021-09-07:
@riya1407 if you look at this line:

```
-- GUI is not enabled
```

CMake is not able to find some dependency to build the GUI code.
Make sure you have run the latest `./etc/DependencyInstaller.sh -dev` and that no errors were reported.
Also, note what maliberty have said:

> Note that you need to have Qt5 installed; Qt4 won't work.

Until the cmake or the `Build.sh -clean` command report `GUI enabled` GUI will not work.

riya1407, 2021-09-07:
what if while running DependencyInstaller.sh we get the following error:
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission
denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches
(13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches
(13: Permission denied)

vvbandeira, 2021-09-07:
@riya1407 that is not ideal -- at least.

Did you run with `sudo` or with `root` permission?

Also, for debug, purposes could you add `set -x` to the `DependencyInstaller.sh` script? This way we can know which command is failing -- my guess is `apt` due to lack of root permission.

riya1407, 2021-09-07:
i had just run shell script by ./DependencyInstaller.sh will it require
root privilege ??

vvbandeira, 2021-09-07:
DependencyInstaller does require root privilege -- either run with sudo or as the root user.
See: https://openroad.readthedocs.io/en/latest/user/BuildLocally.html

riya1407, 2021-09-07:
i tried building with the same command yet showing the above given error
what shall i do?

maliberty, 2021-09-07:
Where is Qt installed on your machine?  You can try in your build area

cmake -DCMAKE_FIND_DEBUG_MODE=TRUE

This will give more info about where it is searching for Qt.

maliberty, 2021-11-28:
This has gone dark.

closed 2021-11-28 by maliberty