The-OpenROAD-Project/OpenROAD #1064: error while running detailed_route

riya1407, 2021-09-07:
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 100 pins.
[INFO DRT-0076]   Complete 200 pins.
[ERROR DRT-0074] No ap for PIN/clk.
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0074
Aborted (core dumped)

where am i going wrong if the above shown content is shown at the console window and openroad exits ??

maliberty, 2021-09-07:
The router can find no way to legally access the top level clk pin.  I would guess that this is because the pin is unplaced.  If not, please provide a test case.

riya1407, 2021-09-07:
may i know the command for the same ??

maliberty, 2021-09-07:
Just look at the input DEF file's PINS section and see if the pin is marked PLACED.

riya1407, 2021-09-07:
may i know the command  to read track file??

maliberty, 2021-09-07:
It is the DEF file I mentioned not a 'track file'.   You would look at results/<platform>/<design>/base/4_cts.def and search for that pin in the PINS section.

Are you running your own design or one of the ones we provide? Are you using OpenROAD-flow-scripts?

riya1407, 2021-09-07:
> results///base/4_cts.def and search for that pin in the PINS section.

solved

closed 2021-09-07 by maliberty