The-OpenROAD-Project/OpenROAD #1065: OpenRoad version / git describe not found

arlpetergadfort, 2021-09-07:
After switching to the git describe version, all I see get is:
-- OpenROAD version: 1
-- OpenROAD git describe: -128-NOTFOUND

maliberty, 2021-09-07:
What happens if you run 'git describe'?

arlpetergadfort, 2021-09-07:
```
git describe
fatal: No names found, cannot describe anything.
```
This happens when I call it from the Openroad-flow / tools / OpenROAD (a submodule)

but when I run it on a non-submodule, I get ``v2.0-854-ga462d24d3``

arlpetergadfort, 2021-09-07:
@maliberty running ``cmake`` on the non-submodule gives me:
```
-- OpenROAD version: 1
-- OpenROAD git describe: v2.0-856-g13d06cda0
```
It seems strange as well that the git describe is saying v2.0 and OR version is 1.

arlpetergadfort, 2021-09-07:
I think I narrowed it down a little. Since we have a copy of the repo, the tags were not copied over. Doing that fixed the  issue with `` -128-NOTFOUND``, but a better fallback or warning might be good.

maliberty, 2021-09-13:
Fixed with 6e80ae096c9ae61e86057b7e9116bc9b44dde37a

closed 2021-09-13 by maliberty