The-OpenROAD-Project/OpenROAD #1068: Building Python bindings as standalone C extension module?

nmoroze, 2021-09-07:
I've been experimenting with the OpenDB Python bindings and ran into an issue. After building OpenROAD, I tried to follow the directions in the OpenDB README (with paths updated), and ran the following Python code:

```python
import importlib.util
spec = importlib.util.spec_from_file_location("odbpy", "./build/src/odb/src/swig/python/odbpy.py")
odb = importlib.util.module_from_spec(spec)
spec.loader.exec_module(odb)
```

However, the final line fails with the error:
```
Traceback (most recent call last):
  File "./build/src/odb/src/swig/python/odbpy.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named '_odbpy'
```

I noticed that the bindings get compiled to a static library `_odbpy.a`, and I believe the issue is that Python looks for C extension modules compiled as shared objects instead. I think what's going on is this static library is used to build the Python shell `odbpy` with the bindings integrated, but the README instructions seem to imply that there should be a C extension module built in this directory as well.

I'm wondering if there's any intent to build a standalone C extension module as well, so that I could access the bindings in a standard Python interpreter (or is there a way to do this already that I'm missing)?

maliberty, 2021-09-07:
We have been moving away from the standalone mode and suggest people use openroad -python to get access to the binding.

maliberty, 2021-09-07:
@vvbandeira we should update the odb README.

nmoroze, 2021-09-09:
Got it, thanks for the quick reply!

closed 2021-09-13 by vvbandeira