The-OpenROAD-Project/OpenROAD #1069: RePlAce diverged at newStepLength

getziadz, 2021-09-08:
Error during `run_placement`

```
[NesterovSolve] Iter: 2270 overflow: 1.62017 HPWL: 295349210
[NesterovSolve] Iter: 2280 overflow: 1.62017 HPWL: 295349210
[NesterovSolve] Iter: 2290 overflow: 1.62017 HPWL: 295349210
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: or_replace.tcl, 93 GPL-0305
```

Design can be found at https://github.com/bsg-idea/bsg_sky130_designs/tree/partial_design_debug/designs/partial_bp_srams

The design is a simple module that contains hardened SRAMs

maliberty, 2021-09-08:
If you are using openroad-flow-script then please use

make global_place_issue

to package the testcase.  Otherwise please provide a self-contained tcl file to reproduce the file.

getziadz, 2021-09-08:
The problem went away after using a custom (larger) DIE_AREA configuration.

closed 2021-09-08 by getziadz