The-OpenROAD-Project/OpenROAD #1072: test failure: incremental02

yurivict, 2021-09-09:
```
Differences found at line 4533.
    - _22166_ INV_X2 + PLACED ( 1232614 897853 ) N ;
    - _22166_ INV_X2 + PLACED ( 1233100 897836 ) N ;
```

```
*** /usr/ports/cad/openroad/work/OpenROAD-d1c7001/src/gpl/test/incremental02.ok	Wed Sep  8 10:52:20 2021
--- /usr/ports/cad/openroad/work/OpenROAD-d1c7001/src/gpl/test/results/incremental02.log	Thu Sep  9 03:55:26 2021
***************
*** 76,112 ****
  [NesterovSolve] Iter: 110 overflow: 0.303736 HPWL: 1210992608
  [NesterovSolve] Iter: 120 overflow: 0.303315 HPWL: 1211084817
  [NesterovSolve] Iter: 130 overflow: 0.302919 HPWL: 1211304054
! [NesterovSolve] Iter: 140 overflow: 0.302559 HPWL: 1211681397
! [NesterovSolve] Iter: 150 overflow: 0.30212 HPWL: 1212462888
! [NesterovSolve] Iter: 160 overflow: 0.302303 HPWL: 1214219394
! [NesterovSolve] Iter: 170 overflow: 0.303145 HPWL: 1217833807
! [NesterovSolve] Iter: 180 overflow: 0.304636 HPWL: 1224580905
! [NesterovSolve] Iter: 190 overflow: 0.3073 HPWL: 1236194664
! [NesterovSolve] Iter: 200 overflow: 0.310978 HPWL: 1254126184
! [NesterovSolve] Iter: 210 overflow: 0.315686 HPWL: 1276503431
! [NesterovSolve] Iter: 220 overflow: 0.319261 HPWL: 1294426202
! [NesterovSolve] Iter: 230 overflow: 0.318325 HPWL: 1290658222
! [NesterovSolve] Iter: 240 overflow: 0.312145 HPWL: 1261520903
! [NesterovSolve] Iter: 250 overflow: 0.309755 HPWL: 1257129419
! [NesterovSolve] Iter: 260 overflow: 0.314495 HPWL: 1292495320
! [NesterovSolve] Iter: 270 overflow: 0.307377 HPWL: 1278866918
! [NesterovSolve] Iter: 280 overflow: 0.301069 HPWL: 1286213629
! [NesterovSolve] Iter: 290 overflow: 0.28998 HPWL: 1286058726
! [NesterovSolve] Iter: 300 overflow: 0.279963 HPWL: 1291567056
! [NesterovSolve] Iter: 310 overflow: 0.266991 HPWL: 1300844487
! [NesterovSolve] Iter: 320 overflow: 0.249807 HPWL: 1309470521
! [NesterovSolve] Iter: 330 overflow: 0.229588 HPWL: 1323578791
! [NesterovSolve] Iter: 340 overflow: 0.212459 HPWL: 1333213614
! [NesterovSolve] Iter: 350 overflow: 0.200486 HPWL: 1326228679
! [NesterovSolve] Finished with Overflow: 0.198316
  [INFO GPL-0133] Unlocked instances
! [NesterovSolve] Iter: 360 overflow: 0.258303 HPWL: 1235297820
! [NesterovSolve] Iter: 370 overflow: 0.223152 HPWL: 1250371582
! [NesterovSolve] Iter: 380 overflow: 0.197022 HPWL: 1258791623
! [NesterovSolve] Iter: 390 overflow: 0.172853 HPWL: 1266498851
! [NesterovSolve] Iter: 400 overflow: 0.153297 HPWL: 1273073518
! [NesterovSolve] Iter: 410 overflow: 0.136149 HPWL: 1278618536
! [NesterovSolve] Iter: 420 overflow: 0.121513 HPWL: 1283025969
! [NesterovSolve] Iter: 430 overflow: 0.10785 HPWL: 1286691488
! [NesterovSolve] Finished with Overflow: 0.099087
! No differences found.
--- 76,114 ----
  [NesterovSolve] Iter: 110 overflow: 0.303736 HPWL: 1210992608
  [NesterovSolve] Iter: 120 overflow: 0.303315 HPWL: 1211084817
  [NesterovSolve] Iter: 130 overflow: 0.302919 HPWL: 1211304054
! [NesterovSolve] Iter: 140 overflow: 0.302559 HPWL: 1211681585
! [NesterovSolve] Iter: 150 overflow: 0.30212 HPWL: 1212462878
! [NesterovSolve] Iter: 160 overflow: 0.302303 HPWL: 1214219693
! [NesterovSolve] Iter: 170 overflow: 0.303145 HPWL: 1217834025
! [NesterovSolve] Iter: 180 overflow: 0.304636 HPWL: 1224581054
! [NesterovSolve] Iter: 190 overflow: 0.3073 HPWL: 1236194629
! [NesterovSolve] Iter: 200 overflow: 0.310979 HPWL: 1254124326
! [NesterovSolve] Iter: 210 overflow: 0.315749 HPWL: 1276519381
! [NesterovSolve] Iter: 220 overflow: 0.319252 HPWL: 1294431982
! [NesterovSolve] Iter: 230 overflow: 0.318365 HPWL: 1290690041
! [NesterovSolve] Iter: 240 overflow: 0.312195 HPWL: 1261542703
! [NesterovSolve] Iter: 250 overflow: 0.309777 HPWL: 1257156346
! [NesterovSolve] Iter: 260 overflow: 0.314441 HPWL: 1292515088
! [NesterovSolve] Iter: 270 overflow: 0.307343 HPWL: 1278870618
! [NesterovSolve] Iter: 280 overflow: 0.300993 HPWL: 1286214041
! [NesterovSolve] Iter: 290 overflow: 0.289899 HPWL: 1286046877
! [NesterovSolve] Iter: 300 overflow: 0.279878 HPWL: 1291570540
! [NesterovSolve] Iter: 310 overflow: 0.266941 HPWL: 1300846816
! [NesterovSolve] Iter: 320 overflow: 0.249739 HPWL: 1309467015
! [NesterovSolve] Iter: 330 overflow: 0.229515 HPWL: 1323585384
! [NesterovSolve] Iter: 340 overflow: 0.212419 HPWL: 1333090688
! [NesterovSolve] Iter: 350 overflow: 0.200219 HPWL: 1326612955
! [NesterovSolve] Finished with Overflow: 0.198269
  [INFO GPL-0133] Unlocked instances
! [NesterovSolve] Iter: 360 overflow: 0.258529 HPWL: 1235269651
! [NesterovSolve] Iter: 370 overflow: 0.223276 HPWL: 1250299299
! [NesterovSolve] Iter: 380 overflow: 0.197196 HPWL: 1258827112
! [NesterovSolve] Iter: 390 overflow: 0.17282 HPWL: 1266529334
! [NesterovSolve] Iter: 400 overflow: 0.153318 HPWL: 1273132466
! [NesterovSolve] Iter: 410 overflow: 0.13593 HPWL: 1278617834
! [NesterovSolve] Iter: 420 overflow: 0.121174 HPWL: 1283051341
! [NesterovSolve] Iter: 430 overflow: 0.107756 HPWL: 1286790796
! [NesterovSolve] Finished with Overflow: 0.099674
! Differences found at line 4533.
!     - _22166_ INV_X2 + PLACED ( 1232614 897853 ) N ;
!     - _22166_ INV_X2 + PLACED ( 1233100 897836 ) N ;
```

rev. d1c7001

clang-11
FreeBSD 13

yurivict, 2021-09-10:
Logs: https://people.freebsd.org/~yuri/openroad-gpl-results.tgz

maliberty, 2021-11-28:
There is no plan for OR to support FreeBSD.  We welcome patches but can't spend the resources to track down issues.

closed 2021-11-28 by maliberty