The-OpenROAD-Project/OpenROAD #1110: "Maze Ripup Wrong"

donn, 2021-09-17:
This issue surfaced when we merged a new version of OpenROAD.

```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical       156045        146237          6.29%
met1       Horizontal     208060        168151          19.18%
met2       Vertical       156045        155652          0.25%
met3       Horizontal     104030        103200          0.80%
met4       Vertical        62418         61626          1.27%
met5       Horizontal      20806         20500          1.47%
---------------------------------------------------------------

[ERROR GRT-0122] Maze ripup wrong.
Error: or_replace.tcl, 93 GRT-0122
```

What is this, exactly?

[aes_cipher.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7187675/aes_cipher.tar.gz)

tspyrou, 2021-09-17:
@eder-matheus can you help Donn from the OpenLane team with this.
@donn Eder is the owner of the global router.

donn, 2021-09-18:
Some more testing notes: this surfaced somewhere between fb8ae93b6c7a5eb0e6fac83360a8a48d76c41885 and f7a44b69256c0b2a1dce50362997c4091d7f6a06.

This passes fine on fb8ae93b6c7a5eb0e6fac83360a8a48d76c41885.

eder-matheus, 2021-09-20:
@donn Could you try the latest commit in the master branch? I've merged a PR yesterday that should fix this issue.

donn, 2021-09-20:
I'll give it a shot

donn, 2021-09-20:
Can confirm that this passed as of 4f4b6da2dd6b740454141f677bbe557d5ef67756.

Thank you for your hard work.

closed 2021-09-20 by donn