The-OpenROAD-Project/OpenROAD #1111: Segfault at global route stage

donn, 2021-09-18:
- OpenROAD Commit: f7a44b69256c0b2a1dce50362997c4091d7f6a06 (Tested with 01461c345522aa9cc240df2d7f01c0cd886f23d0 outside OpenLane)

- Keeps spamming this then crashes:
```
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 2 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 2 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 2 T shapes.
```

- Reproducible: [yhuff_sigvio.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7190183/yhuff_sigvio.tar.gz)

maliberty, 2021-09-18:
@donn are you sure its from global_route and not repair_antennas ?

donn, 2021-09-18:
Let me verify...

donn, 2021-09-18:
Oh, you're right. Oops.

Well I guess a workaround would be to switch the diode insertion strategy in OpenLane. Still, shouldn't segfault...

maliberty, 2021-09-22:
This fails deep in orderWires, an awful piece of code in odb.

donn, 2021-09-23:
Oof. I'm so sorry.

kammoh, 2021-11-12:
I think I'm hitting the same issue. Present in current HEAD (c07e55dd4). Tried going back a little bit but still present.
I can reproduce it with OpenLane's `aes` design, on macOS, when running `scripts/openroad/or_groute.tcl`. I've experience a similar segfault on a fedora machine as well, but need to verify if it was the same thing.

Here's the stack trace:

```c
[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1               3932           217            5.52%             0 /  0 /  0
met1            386532        108103           27.97%             0 /  0 /  0
met2            355124        112185           31.59%             0 /  0 /  0
met3            235904          3998            1.69%             0 /  0 /  0
met4            140476            15            0.01%             0 /  0 /  0
met5             46816             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total          1168784        224518           19.21%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 2053502 um
[INFO GRT-0006] Repairing antennas, iteration 1.
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0224] Chose via L1M1_PR as default.
[INFO GRT-0224] Chose via M1M2_PR as default.
[INFO GRT-0224] Chose via M2M3_PR as default.
[INFO GRT-0224] Chose via M3M4_PR as default.
[INFO GRT-0224] Chose via M4M5_PR as default.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Process 14473 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x38)
    frame #0: 0x00000001003e087c openroad`odb::tmg_conn::treeReorder(this=0x000000012cefc400, verbose=false, quiet=<unavailable>, no_convert=false) at tmg_conn.cpp:2015:32 [opt]
   2012	    }
   2013	    if (pt) {
   2014	      tstack0--;
-> 2015	      _last_id = x->_first_pt->_dbwire_id;
   2016	      if (verbose) {
   2017	        notice(0, "net %d feedthru at ", _net->getId());
   2018	        print_rcterm(x);
Target 0: (openroad) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x38)
  * frame #0: 0x00000001003e087c openroad`odb::tmg_conn::treeReorder(this=0x000000012cefc400, verbose=false, quiet=<unavailable>, no_convert=false) at tmg_conn.cpp:2015:32 [opt]
    frame #1: 0x00000001003e017c openroad`odb::tmg_conn::analyzeNet(this=0x000000012cefc400, net=0x000000010b845c80, force=<unavailable>, verbose=false, quiet=false, no_convert=<unavailable>, cutLength=<unavailable>, maxLength=<unavailable>, no_patch=true) at tmg_conn.cpp:1752:5 [opt]
    frame #2: 0x00000001003e7d98 openroad`odb::orderWires(net=0x000000010b845c80, force=false, verbose=false) at wOrder.cpp:132:10 [opt]
    frame #3: 0x0000000100645644 openroad`grt::AntennaRepair::checkAntennaViolations(this=<unavailable>, routing=<unavailable>, max_routing_layer=<unavailable>, diode_mterm=0x00000001055df778) at AntennaRepair.cpp:122:7 [opt]
    frame #4: 0x000000010064e15c openroad`grt::GlobalRouter::repairAntennas(this=0x000000010310f860, diode_port=<unavailable>, iterations=3) at GlobalRouter.cpp:237:37 [opt]
    frame #5: 0x0000000100643ea0 openroad`_wrap_repair_antennas(void*, Tcl_Interp*, int, Tcl_Obj* const*) [inlined] grt::repair_antennas(diodePort=0x0000600003e3cb00, iterations=<unavailable>) at GlobalRouterTCL_wrap.cxx:1831:22 [opt]
    frame #6: 0x0000000100643e90 openroad`_wrap_repair_antennas(clientData=<unavailable>, interp=0x0000000109818210, objc=<unavailable>, objv=<unavailable>) at GlobalRouterTCL_wrap.cxx:2667:7 [opt]
    frame #7: 0x0000000102ea3ea4 libtcl8.6.dylib`TclNRRunCallbacks + 80
    frame #8: 0x0000000102ea4e98 libtcl8.6.dylib`TclEvalEx + 1616
    frame #9: 0x0000000102ea57bc libtcl8.6.dylib`Tcl_Eval + 40
    frame #10: 0x0000000100016304 openroad`ord::tclAppInit(Tcl_Interp*) [inlined] tclReadlineInit(interp=0x0000000109818210) at Main.cc:236:22 [opt]
    frame #11: 0x0000000100016264 openroad`ord::tclAppInit(Tcl_Interp*) [inlined] tclAppInit(argc=<unavailable>, argv=<unavailable>, interp=<unavailable>) at Main.cc:349:12 [opt]
    frame #12: 0x0000000100015f00 openroad`ord::tclAppInit(interp=0x0000000109818210) at Main.cc:358:10 [opt]
    frame #13: 0x0000000102f4c658 libtcl8.6.dylib`Tcl_MainEx + 668
    frame #14: 0x0000000102f4d0c0 libtcl8.6.dylib`Tcl_Main + 52
    frame #15: 0x000000010001677c openroad`main(argc=<unavailable>, argv=0x000000016fdff170) at Main.cc:217:3 [opt]
```
as you can see `x->_first_pt` is null.

maliberty, 2021-11-12:
A fix for a very similar issue just went in (https://github.com/The-OpenROAD-Project/OpenROAD/pull/1293)  Please update and try again.

closed 2021-12-22 by donn

QuantamHD, 2021-12-22:
This issue is a duplicate of #1428