The-OpenROAD-Project/OpenROAD #1133: `write_sdf` missing delay values for the typical corner

Manarabdelaty, 2021-09-24:
The sdf file output from `write_sdf` doesn't have delay values for the typical corner. It only has the minimum and the maximum corner. For example, this path only has the max and min delays
`    (IOPATH A X (0.069::0.128) (0.069::0.132))
`

I tried to explicitly define the corners and read the liberty with the `-corner typ` option, but it didn't help.  I am attaching a test case to reproduce the issue.
[spm_test_case.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/7225806/spm_test_case.zip)

maliberty, 2021-10-19:
sdf does not require all three corners to be present and two is quite typical.  What is the simulator issue?

Manarabdelaty, 2021-10-19:
The issue is that the skywater-pdk has a placeholder for the three corners in the specify blocks, so the simulator expects to see three corners in the sdf file as well.

jjcherry56, 2021-11-09:
There is no relationship between process corner names and the position a delay is written in an sdf triple.
The min/max values written by opensta are specifically designed to be compatible with static timing analysis
and because sdf only supports 3 values they are the min/max values for a specific process corner.
I added an option -include_typ that inserts the meaningless average of min and max just to make the confused
simulator happy.

closed 2021-11-09 by jjcherry56