The-OpenROAD-Project/OpenROAD #1136: How to fix "RePlAce divergence detected."problem?

shuolai, 2021-09-27:
I met a problem in the 3_1 global place stage and now sure how to solve it. I tried to increase the core_area but it doesn't work. 
What does the "max_phi_cof value" mean? 
[image]

maliberty, 2021-09-27:
global_placement has a -max_phi_coef but it is rather obscure though you can try different values.

What happened when increased the core area?  Did the overflow reach a lower level or was it the same 0.136?  Can you provide a test case?

shuolai, 2021-09-29:
Hi, Matt. I just found that I didn't place macros before the global_place. I suppose this can be the reason of this problem? 
The following warning messages appear in global_placement stage.
[image]

maliberty, 2021-09-29:
It might be related but you need to fix it in any case.  The messages suggest you have the wrong CLASS in the lef for your macros (it should be BLOCK).  Replace will work around it but the macro placer doesn't so you need to fix it.

tcal-x, 2022-09-19:
I am hitting this same error on a very small design.   There are no warnings in the log.

It's a very frustrating message `Re-run with a smaller max_phi_cof value` when it

1. doesn't tell you how to change the value
2. doesn't tell you what the current / default / previously-used value was, so you have no idea what value to try next.

maliberty, 2022-09-20:
Its a pretty poor message.  Can you provide a test case as I would rather see if we can fix it directly rather than having users mess with internal knobs.

sharkoo7, 2022-09-21:
I came across the same error with this test case. I tried to change `global_place_density` and
`global_place_density`.But it doesn't work.
[test_case.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9612739/test_case.zip)

tcal-x, 2022-09-22:
Hi @maliberty , thanks for the reply.   I was able to work around this issue using Matt Venn's advice for ANY problem that you hit with a very small design -- add this to the design's config.tcl:
```
set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 200 200"
```

My example uses a custom synth.tcl script, so it is a little more complicated than you might expect, but I can still send it if you think it's useful.  Perhaps not necessary since @sharkoo7 has already provided a testcase?

maliberty, 2022-09-22:
@ramezmoussa in the past I've seen problems where the placer confuses very fast convergence on trivial designs with divergence.    This will be a good chance to learn the convergence code.  The message is stale and unhelpful so please rework that as well.

vijayank88, 2023-06-14:
@maliberty 
This is not having right test case. Still wants to keep open?