The-OpenROAD-Project/OpenROAD #1146: [OpenFASoC] write_cdl issue

shuolai, 2021-10-01:
The temp_sensor in OpenFASoC has LVS issue after the update of write_cdl.

Link to OpenFASoC:  (https://github.com/idea-fasoc/OpenFASOC)

The power nets in cdl file are unconnected.
[image]

I also checked the 6_final.v file and found that the power nets disappeared when using the latest version of the OpenROAD compared with the older version. I think maybe this could be a reason causing the error in cdl file? 
[image]

maliberty, 2021-10-01:
Please package a test case

msaligane, 2021-10-02:
@maliberty Thanks for taking a look. This issue has been dragging for 2 weeks now :/

initially we made a [script ](https://github.com/idea-fasoc/OpenFASOC/blob/e055fed0a9401733907a1cd07457d4331187c920/generators/temp-sense-gen/flow/util/cdl_parser.py) to fix the write_cdl command errors (I think it was the orders of the pins that was messed up. Then just before running LVS using netgen we would run cdl_parser.py on the .cdl generated by OR as follows:

```
netgen_lvs: $(RESULTS_DIR)/6_final.gds
	python util/cdl_parser.py -i $(RESULTS_DIR)/6_final.cdl -s $(PLATFORM_DIR)/cdl/sky130_fd_sc_hd.spice -l $(OBJECTS_DIR)/merged_spacing.lef -o $(DESIGN_NAME).spice
	$(COMMON_VERIF_DIR)/run_lvspex.sh $(RESULTS_DIR)/6_final.gds $(DESIGN_NAME) $(REPORTS_DIR)/6_final_lvs.rpt
```
link to [Makefile](https://github.com/idea-fasoc/OpenFASOC/blob/e055fed0a9401733907a1cd07457d4331187c920/generators/temp-sense-gen/flow/Makefile#L491-L493).

See attached [tempsense.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/7270799/tempsense.tar.gz).
I think based on what I have checked (2weeks ago) I think there is an issue with the power/wells pins on the second voltage domains. You can check in the tar ball:
 
- Flow scripts based on ORFS: `/tempsense/generators/temp-sense-gen/flow`
- lvs report from netgen:  `/tempsense/generators/temp-sense-gen/flow/reports/sky130hd/tempsense/6_final_lvs.rpt`
- verilog / cdl /def files: `/tempsense/generators/temp-sense-gen/flow/results/sky130hd/tempsense`

Please let me know if you have any questions and thanks again!

maliberty, 2021-10-02:
What is CDL_FILE set to?  I don't see anything setting it in your tgz.

maliberty, 2021-10-02:
I also see many warnings like:

[WARNING ODB-0286] Terminal VNB of CDL master sky130_fd_sc_hd__a2111o_1 not found in LEF.

which suggests cdl/lef mismatches.

msaligane, 2021-10-02:
> What is CDL_FILE set to? I don't see anything setting it in your tgz.
`export CDL_FILE                 = ../blocks/$(PLATFORM)/cdl/sky130_fd_sc_hd_merged.cdl
`
it is in the `config.mk` in `/tempsense/generators/temp-sense-gen/flow/design/sky130hd/tempsense`

> [WARNING ODB-0286] Terminal VNB of CDL master sky130_fd_sc_hd__a2111o_1 not found in LEF.

Then it is probably a source file issue. @shuolai please double check the LEF/CDL files versus the one in latest of open_pdk. If it is still there please file an issue.

msaligane, 2021-10-02:
@maliberty Why do we need the LEF to write the cdl. I think verilog and cdl primitives should be the one to be used?

maliberty, 2021-10-02:
The database is loaded from LEF/DEF in every step after the first.

msaligane, 2021-10-02:
> The database is loaded from LEF/DEF in every step after the first.
Sure but we could use `6_final.v` file which is the one we would eventually use if wanted to the a verilog to lvs cdl conversion. Right?

maliberty, 2021-10-02:
6_final.v is just generated from the DEF file (there is only read_def in final_report.tcl).  You won't get any extra info there.

maliberty, 2021-11-13:
@msaligane is this still relevant?

msaligane, 2021-11-13:
@maliberty Yes, this is still an issue. I have tried using the same LEF but still seeing weird results during LVS. Have you been able to validate the `write_cdl ` update in any of the other pdks we have?

I have asked @erictaur and @wenbodd to take a look but I did not hear back from them yet.

maliberty, 2021-11-13:
@msaligane if you have an updated test with a LEF I can take a look.  I don't know what "weird results" mean.

msaligane, 2021-11-13:
@maliberty That would be really helpful. Weird results in LVS means that the VNB and VPB are named as unrecognized net in the 2nd voltage domain of the sensor.

@erictaur do you mind sharing the latest details since you have met with @wenbodd ?
Have you modified the LEF as discussed ? 

The issue is packaged above. I will try to go over this again on Monday afternoon.

maliberty, 2021-11-13:
@msaligane in the packaged test case the lef & cdl did not align.  Until that is addressed I don't have anything to look at.

msaligane, 2021-11-30:
@maliberty Actually if you look at the LEF files in the tar  I attached. VNB / VPB are there in the CDL and LEF files. 
`sky130_fd_sc_hd__a2111o_1 `  this cell is not used in the design at all.

maliberty, 2021-11-30:
Here is a master you are using (it's the same):
[WARNING ODB-0286] Terminal VPB of CDL master sky130_fd_sc_hd__inv_1 not found in LEF.

In the CDL:
.SUBCKT sky130_fd_sc_hd__inv_1 A VGND VNB VPB VPWR Y

but in the LEF:
MACRO sky130_fd_sc_hd__inv_1
  PIN A
  PIN Y
  PIN VGND
  PIN VPWR
END sky130_fd_sc_hd__inv_1

So the CDL & LEF mismatch as OR reports.  There is no way for OR to know what to write for VNB & VPB since they aren't in the LEF (or DEF).  It's a data problem not a tool problem.

maliberty, 2021-11-30:
In the LEF in openlane I do see the bulk pins.  I think ORFS needs a refresh.

msaligane, 2021-11-30:
There are two types of LEF, [magic.lef](https://cs.opensource.google/skywater-pdk/skywater-pdk/+/0e1fa89b2db152cbdc75c8b1e20cbfecba118b8d:libraries/sky130_fd_sc_hd/latest/cells/inv/sky130_fd_sc_hd__inv_1.magic.lef) and [.lef](https://cs.opensource.google/skywater-pdk/skywater-pdk/+/0e1fa89b2db152cbdc75c8b1e20cbfecba118b8d:libraries/sky130_fd_sc_hd/latest/cells/inv/sky130_fd_sc_hd__inv_1.lef;dlc=05eddae23a77687315df2783c850628f80248535)
The magic.lef is the one which has VPB/VNB defined in it. The original version .lef did not have those pins ( I do not understand why there is a need to define those pins in an inverter - I might be missing something).

@minghungumich Do you think you could update to the LEFs straight out of `open_pdks`. I think right now, we are using the one `ORFS/platforms`.

maliberty, 2021-11-30:
@mithro @RTimothyEdwards can you explain why [hd](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/main/cells/inv/sky130_fd_sc_hd__inv_1.lef) and [hs](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hs/blob/main/cells/inv/sky130_fd_sc_hs__inv_1.lef) versions of this cell have different handling of bulk pins?  hd doesn't have them while hs does.

mithro, 2021-11-30:
@maliberty - I assume because different developers created the libraries?

RTimothyEdwards, 2021-11-30:
@maliberty :  I checked and the differences in the handling of the pins goes back to the original SkyWater files, so I second Tim Ansell's opinion.

mithro, 2021-11-30:
It sounds like it **might** make more sense to use the generated `.magic.lef` files?

RTimothyEdwards, 2021-11-30:
@mithro :  No, there is no information about the location of pins other than what's in the original files, so the same issue ends up in all formats of all files no matter what tool generated or modified them.  So the pins will be in different places in the different standard cell libraries unless somebody modifies them all specifically to make them mutually compatible.

mithro, 2021-11-30:
I thought the issue is that the bulk pins exist in the `hd` cell's `.lef` files but **_not_** in the `hs` cell's `.lef` files?

@maliberty said above;
> `hd` **doesn't** have them while `hs` **does**.

> In the `.magic.lef` they are present in **both**.

msaligane, 2021-11-30:
I think adding those pins in cells that do not have explicit well connection on them is confusing and creates a bunch of issues. However, if we want to add them for some reason, then we should have consistency since we get a lot lvs issues.

maliberty, 2021-11-30:
I discussed bulk pins with @louiic who recommends we have them in cdl & lef (it is required at lower nodes so it is best to have a single methodology to support).

RTimothyEdwards, 2021-11-30:
@mithro :  If pins are not present in all files, then somebody is not using the approved set of files from the open_pdks installation.  In the open_pdks libraries, all cell views declare all four power pins.  There are still differences between the libraries as to how the pins are positioned, which what what I thought was the issue here.

@msaligane :  On the contrary, it is necessary to define the explicit well connections for the LEF to represent the correct connectivity.  All of the other file types (CDL, GDS, verilog, etc.) declare 4 power pins for every cell.  If the LEF file does not also include four power pins, then various tools start complaining about it.

RTimothyEdwards, 2021-11-30:
@msaligane :  Also, if you're using netgen for LVS, then why not just compare the extracted layout against the verilog?

msaligane, 2021-11-30:
> I discussed bulk pins with @louiic who recommends we have them in cdl & lef (it is required at lower nodes so it is best to have a single methodology to support).
I think it has to do with the LEF verison (IE 58 here)

> @msaligane : On the contrary, it is necessary to define the explicit well connections for the LEF to represent the correct connectivity. All of the other file types (CDL, GDS, verilog, etc.) declare [...]
Sounds good, so maybe let's go with this plan. I am happy with either. Who can update the files? maybe in the precompiled version of `open_pdks`?

RTimothyEdwards, 2021-11-30:
@msaligane :  Yes, I think the origin of the whole problem was that the LEF files we got from SkyWater were < version 5.7, which is the spec version that introduced the nwell/pwell masterslice layers and allowed them to connect to pins.  So with the earlier version of the LEF spec, they did all sorts of bizarre things like putting the VPB and VNB pins on metal layers where they don't actually exist.  I ended up getting magic to pull the relevant information about pin use/direction/shape out of the LEF file, discarding everything else, then regenerating the LEF from scratch, which helps make all the files for all the standard cell libraries look reasonably similar.

maliberty, 2021-11-30:
The ORFS versions don't have the bulk pins so it sounds like it is time for an update.

maliberty, 2021-11-30:
Does OL use the .magic.lef or the .lef today?

msaligane, 2021-11-30:
@maliberty I think we can update the platform to `magic.lef` which I think it is what is used by OL.

@RTimothyEdwards makes sense! One issue though is that for teaching classes next semester, we use commercial tools and the file have multiple issues when they are used related to  VPB/VNB.

RTimothyEdwards, 2021-12-01:
@msaligane :  (1) Note that "use the open_pdks installation" is not the same as "use the magic.lef".  The files in the skywater-pdk repository were made prior to a number of updates and fixes in magic's LEF generation.  For the most part, all the LEF files in the skywater-pdk repository are wrong in some way, whether they come from the original sources or from magic.  (2) For the same reason, the LEF files produced by the open_pdks installation have gone through a number of bug fixes, and as long as the commercial tools understand LEF format 5.7, they should no longer cause issues.

maliberty, 2021-12-01:
@mithro can we upstream these fixes to skywater-pdk?

mithro, 2021-12-01:
@maliberty - Yes but there is no timeline for that happening. Soe for now you should use the output produced by `open_pdks`.

mithro, 2021-12-01:
BTW -- I'm pretty sure this is what OpenLane is using.

maliberty, 2021-12-02:
I am syncing ORFS to OL's open_pdk sky130hd LEFs (tech & stdcell) in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/243

maliberty, 2021-12-02:
The above is merged - @shuolai  @msaligane can we close this?

msaligane, 2021-12-02:
@maliberty we need to test it first. I don't think @minghungumich  got to it yet. I will try to run this in the next 2 days.

vijayank88, 2023-06-14:
@msaligane 
Can you please update the status?
If resolved mean close this issue.

msaligane, 2023-06-16:
@alibillalhammoud @Wen-Tian-Pineapple
Can you guys double-check this? I think it is resolved now.

stevvik, 2023-06-22:
Hi everybody!
I still have the same issue with unconnected instances in cdl. Even in "gcd" test example for sky130hd.
And also in my custom technology. Pins in LEF files correspond to Verilog.
It looks like the problem cells are only instances that were inserted in verilog during optimization (input*, output*, rebuffer* instance names).

maliberty, 2023-06-22:
I ran sky130/gcd and I see:

Xinput1 req_msg[0] VSS VSS VDD VDD net1 sky130_fd_sc_hd__clkbuf_1

which looks to match 

.SUBCKT sky130_fd_sc_hd__clkbuf_1 A VGND VNB VPB VPWR X

what is the issue?

stevvik, 2023-06-22:
@maliberty, when the issue was fixed?
I have the following in cdl using ORFS locally installed in June 8, 2023:
Xinput1 req_msg[0] _unconnected_0 _unconnected_1 _unconnected_2
+ _unconnected_3 net1 sky130_fd_sc_hd__clkbuf_1

maliberty, 2023-06-22:
I just used the head of master but there haven't been any changes in quite a while.  I'm guessing something is wrong upstream.  How are you running this?

stevvik, 2023-06-22:
From flow dir.
Uncomment DESIGN_CONFIG=./designs/sky130hd/gcd/config.mk in Makefile.
Then first run "make", then "make lvs".

stevvik, 2023-06-22:
I got the same using docker installation.

stevvik, 2023-06-22:
@maliberty 
Yes, this helped! Thank you!
But I still have LVS failed for sky130hd "gcd" testcase with the following error:
ERROR: In ./platforms/sky130hd/lvs/sky130hd.lylvs: Pin count mismatch between circuit definition and circuit call: 6 expected, got 7 in <my_long_path>/objects/sky130hd/gcd/base/6_final_concat.cdl, line 11726 in Netlist::read

maliberty, 2023-06-22:
Would you paste the line from the error?

maliberty, 2023-06-22:
I get a totally different error
```
ERROR: In ./platforms/sky130hd/lvs/sky130hd.lylvs: Expected a real number here: normal are .. in /workspaces/mliberty/w8/OpenROAD-flow-scripts/flow/objects/sky130hd/gcd/base/6_final_concat.cdl, line 577 in Netlist::read
```
I'm not sure what to make of it as I don't work on the lvs tool and it looks legal.

stevvik, 2023-06-23:
> Would you paste the line from the error?

XI1 VGND VNB VPB VPWR net59 LO / sky130_fd_sc_hd__conb_1

This is the instance from subckt sky130_fd_sc_hd__macro_sparecell
Error says that this instance contains 7 pins connection. It seems that it treats backslash as pin.
The subckt subckt sky130_fd_sc_hd__macro_sparecell is the only one that contains backslashes in instances.

I deleted backslashes, re-run LVS and got further errors:
ERROR: In ./platforms/sky130hd/lvs/sky130hd.lylvs: Can't find a value for a R, C or L device in /proj/eda/fpi2_digital_flow/dev/stolbikov/work/ORFS/TEST_ORFS/gcd_local/run_2/objects/sky130hd/gcd/base/6_final_concat.cdl, line 5460 in Netlist::read

Need somebody to fix all CDL inconsistences...

stevvik, 2023-06-23:
@maliberty Matt, can you explain me how openroad knows the pins order during write_cdl command?
In my technology in resulted 6_final.cdl I'm getting (for invertor, as an example):
X_192_ net1 net2 VDD VSS INV
Why it placed power pins in the end in order "VDD VSS"?
In LEF MACRO for invertor I have the following pins order "A Z gnd vdd".
In 6_final.v Verilog power pins are absent.

But in sky130hd I see that write_cdl can put power pins between signal pins. How he knows?

rovinski, 2023-06-23:
The `write_cdl` command has a `-masters` argument which takes in the component CDL files. `write_cdl` will match the pin order found in the input files. If the component is not found in the input files, ERROR ODB-287 is thrown.

I noticed from the code that it doesn't look like it checks for mismatches where the CDL has more pins than the LEF. That might be a good check to add.

maliberty, 2023-06-27:
Apparently we allow such https://github.com/The-OpenROAD-Project/OpenROAD/blob/f413f8405db24165e2db0efad5dba4548d85fe7e/src/odb/src/cdl/cdl.cpp#L165

maliberty, 2023-06-27:
The error I see from klayout is due to "topography=normal" in the skywater provided CDL.   I tried removing those locally and then I run into

```
.SUBCKT sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
*.PININFO VGND:I VNB:I VPB:I VPWR:I HI:O LO:O
rI12 VGND LO short
rI11 HI VPWR short
.ENDS sky130_fd_sc_hd__conb_1
```
which gives "ERROR: Can't find a value for a R, C or L device".   That seems a reasonable error.  I don't see how people can use this CDL.

@RTimothyEdwards any ideas?

maliberty, 2024-09-03:
This is approaching three years old and I don't see a clear action to take.

closed 2024-09-03 by maliberty