The-OpenROAD-Project/OpenROAD #1149: Build breaks: too many arguments to function call, expected single argument 'cell', have 2 arguments, etc

yurivict, 2021-10-03:
```
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/dbSta/src/dbNetwork.cc:915:23: error: too many arguments to function call, expected single argument 'cell', have 2 arguments
  groupBusPorts(cell, [](const char*) { return true; });
  ~~~~~~~~~~~~~       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/sta/include/sta/ConcreteNetwork.hh:179:16: note: 'groupBusPorts' declared here
  virtual void groupBusPorts(Cell *cell);
               ^
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/dbSta/src/dbNetwork.cc:950:19: error: cannot initialize a parameter of type 'sta::ConcreteLibrary *' with an lvalue of type 'sta::Library *'
    deleteLibrary(top_lib);
                  ^~~~~~~
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/sta/include/sta/ConcreteNetwork.hh:238:39: note: passing argument to parameter 'library' here
  void deleteLibrary(ConcreteLibrary *library);
                                      ^
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/dbSta/src/dbNetwork.cc:962:17: error: too many arguments to function call, expected single argument 'cell', have 2 arguments
                [=](const char *port_name) { return portMsbFirst(port_name); } );
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/openroad/work/OpenROAD-2.0-1185-gf440b2b59/src/sta/include/sta/ConcreteNetwork.hh:179:16: note: 'groupBusPorts' declared here
  virtual void groupBusPorts(Cell *cell);
               ^
3 errors generated.
```

rev. f440b2b59
clang-11
FreeBSD 13

maliberty, 2021-10-03:
I think your sta submodule is out of date.

yurivict, 2021-10-04:
You are correct. Sorry for bothering you.

closed 2021-10-04 by yurivict